Robotics-STAR-Lab/H2-Mapping #2: Do you have any plan to release this work?

tlok666, 2023-06-27:
Thanks for this interesting work. Would you be able to release the code?

JIANG-CX, 2023-07-14:
> Thanks for this interesting work. Would you be able to release the code?

Thank you for your interest in our work. We plan to release the source code after the paper is accepted. Thank you.

cv-lab-x, 2023-08-17:
> > Thanks for this interesting work. Would you be able to release the code?
> 
> Thank you for your interest in our work. We plan to release the source code after the paper is accepted. Thank you.

Hi, which conferences do you submit?

JIANG-CX, 2023-08-17:
> > > Thanks for this interesting work. Would you be able to release the code?
> > 
> > 
> > Thank you for your interest in our work. We plan to release the source code after the paper is accepted. Thank you.
> 
> Hi, which conferences do you submit?

This paper is submitted to "IEEE Robotics and Automation Letters", and it is currently undergoing a review process.

mattiapiz, 2023-09-15:
Hi, any news? 

> > Thanks for this interesting work. Would you be able to release the code?
> 
> Thank you for your interest in our work. We plan to release the source code after the paper is accepted. Thank you.

Hi, any news? Very interesting work, I can not wait :)

JIANG-CX, 2023-09-15:
Thanks for your interest. The code will be released at the end of this month. Thanks. From: ***@***.*** ***@***.***> on behalf of Mattia Pizzuto ***@***.***>Sent: Friday, September 15, 2023 17:01To: SYSU-STAR/H2-Mapping ***@***.***>Cc: JIANG Chenxing ***@***.***>; Comment ***@***.***>Subject: Re: [SYSU-STAR/H2-Mapping] Do you have any plan to release this work? (Issue #2) 
Hi, any news?

Thanks for this interesting work. Would you be able to release the code?

Thank you for your interest in our work. We plan to release the source code after the paper is accepted. Thank you.

Hi, any news? Very interesting work, I can not wait :)

closed 2023-09-26 by JIANG-CX