Robotics-STAR-Lab/H2-Mapping #7: terminate called without an active exception in vins rgbd

small-zeng, 2023-10-02:
when I run `bash ros_cmd/run_vins_rgbd.sh `  , it occurs:
`[ WARN] [1696257977.955352297]: Reason given for shutdown: [[/rvizvisualisation] Reason: new node registered with same name]
[ INFO] [1696257978.008451233]: Extrinsic_R : 
   0.999969  0.00614047  0.00501018
-0.00594551    0.999258  -0.0380416
-0.00524006   0.0380106    0.999264
[ INFO] [1696257978.008503466]: Extrinsic_T : 
-0.000121045    -0.020129   -0.0219023
[ INFO] [1696257978.008520190]: Synchronized sensors, fix time offset: -0.015
[ WARN] [1696257978.008880801]: Waiting for image and imu...
terminate called without an active exception
[vins_estimator-1] process has died [pid 422, exit code -6, cmd /H2-Mapping/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/zju/.ros/log/ed278d04-612b-11ee-963d-3cecef3dffc0/vins_estimator-1.log].
log file: /home/zju/.ros/log/ed278d04-612b-11ee-963d-3cecef3dffc0/vins_estimator-1*.log
`

JIANG-CX, 2023-10-03:
The error appears to be "Reason: a new node has been registered with the same name." To resolve this, you can attempt using the command "rosnode kill --all" and subsequently restarting.

small-zeng, 2023-10-03:
> The error appears to be "Reason: a new node has been registered with the same name." To resolve this, you can attempt using the command "rosnode kill --all" and subsequently restarting.

Thanks