Robotics-STAR-Lab/H2-Mapping #8: 怎么实时重建？

chenghaozhangswu, 2023-10-07:


JIANG-CX, 2023-10-07:
论文中的实物实验就是使用了“Run in ROS（full SLAM）” 中的代码，谢谢。

chenghaozhangswu, 2023-10-07:
抱歉，我没有看懂，应该如何通过指令进入实时重建，还是说需要提前录制bag才行？

JIANG-CX, 2023-10-07:
实际使用时不需要录制bag，只需要更改topic名字与config文件，使得和你使用的rgbd相机的输入适配就可以。

chenghaozhangswu, 2023-10-07:
> 实际使用时不需要录制bag，只需要更改topic名字与config文件，使得和你使用的rgbd相机的输入适配就可以。

应该从哪里进入呢，是按照哪个步骤进行？

JIANG-CX, 2023-10-07:
流程与Run in ROS（full SLAM）和其中的self-captured house dataset相同，config的设置请参看Run in ROS（full SLAM）的第三点，以及其中use your own rgbd dataset中的注意事项

chenghaozhangswu, 2023-10-07:
> 流程与Run in ROS（full SLAM）和其中的self-captured house dataset相同（只是不需要播bag，而是传感器实时输入），config的设置请参看Run in ROS（full SLAM）的第三点，以及其中use your own rgbd dataset中的注意事项

我使用的也是l515，是使用roslaunch realsense2_camera rs_aligned_depth_lidar.launch &取代播放bag吗？rs_aligned_depth_lidar.launch这里是如何配置的？

JIANG-CX, 2023-10-07:
l515的配置可以参看 https://github.com/IntelRealSense/realsense-ros

chenghaozhangswu, 2023-10-11:
> l515的配置可以参看 https://github.com/IntelRealSense/realsense-ros

能否提供一个用自己l515相机实时扫描的操作步骤参考，谢谢

JIANG-CX, 2023-10-11:
根据readme中Use your own RGB-D sequence修改config文件
根据https://github.com/IntelRealSense/realsense-ros启动realsense
根据readme中Self-captured House dataset的步骤3、4启动mapping和tracking

chenghaozhangswu, 2023-10-11:
> 根据readme中Use your own RGB-D sequence修改config文件 根据https://github.com/IntelRealSense/realsense-ros 启动 realsense 根据readme中Self-captured House dataset的步骤3、4启动mapping和tracking

您好，请问能否提供那个realsense启动文件

JIANG-CX, 2023-10-11:
> > 根据readme中Use your own RGB-D sequence修改config文件 根据https://github.com/IntelRealSense/realsense-ros 启动 realsense 根据readme中Self-captured House dataset的步骤3、4启动mapping和tracking
> 
> 您好，请问能否提供那个realsense启动文件

请根据自己的场景来设置诸如曝光等realsense参数，realsense的文档已经清楚写明了如何启动[Usage Instructions](https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#usage-instructions)，谢谢

chenghaozhangswu, 2023-10-13:
> 请根据自己的场景来设置诸如曝光等realsense参数，realsense的文档已经清楚写明了如何启动[Usage Instructions](https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy#usage-instructions)，谢谢

能提供案例中录制tower_compass.bag的launch文件参考吗？

JIANG-CX, 2023-10-13:
launch 文件：
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="json_file_path"      default="$(find realsense2_camera)/config/L515_config_tower_manual_expo_white_high_bright_2.json"/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="true"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="400"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>
  <arg name="enable_pose"         default="true"/>

  <arg name="enable_pointcloud"   default="false"/>
  <arg name="enable_sync"         default="true"/>
  <arg name="align_depth"         default="true"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-1"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/> 

  <rosparam>
      /camera/motion_module/global_time_enabled: true
      /camera/l500_depth_sensor/global_time_enabled: true
      /camera/rgb_camera/global_time_enabled: true
      
      /camera/rgb_camera/enable_auto_exposure: false
      /camera/rgb_camera/enable_auto_white_balance: false
      
      /camera/rgb_camera/exposure: 100
      /camera/rgb_camera/white_balance: 4500
      /camera/rgb_camera/gain: 4096    
      
      /camera/l500_depth_sensor/min_distance: 100
      /camera/l500_depth_sensor/noise_filtering: 6 

  </rosparam>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="json_file_path"           value="$(arg json_file_path)"/>

      <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
      <arg name="enable_sync"              value="$(arg enable_sync)"/>
      <arg name="align_depth"              value="$(arg align_depth)"/>
[... 61 lines cut ...]
        "Humidity Temperature": 36.1058959960938,
        "Inter Cam Sync Mode": 0.0,
        "Invalidation Bypass": 0.0,
        "LDD temperature": 36.3938369750977,
        "Laser Power": 100,
        "Ma Temperature": 34.7539749145508,
        "Mc Temperature": 35.9714241027832,
        "Min Distance": 250,
        "Noise Estimation": 0.0,
        "Noise Filtering": 6,
        "Post Processing Sharpening": 1,
        "Pre Processing Sharpening": 0.0,
        "Receiver Gain": 9,
        "Sensor Mode": 0.0,
        "Visual Preset": 0.0
    },
    "schema version": 1,
    "viewer": {
        "stream-depth-format": "Z16",
        "stream-fps": "30",
        "stream-height": "480",
        "stream-ir-format": "Y8",
        "stream-width": "640"
    }
}

lvmingzhe, 2023-10-15:
请问L515_config_tower_manual_expo_white_high_bright_2.json能提供给大家参考一下吗？谢谢
主要不知道该设置哪些参数和格式

chenghaozhangswu, 2023-10-16:
> launch 文件：
>
> ```
> <launch>
>   <arg name="serial_no"           default=""/>
>   <arg name="json_file_path"      default="$(find [...]
感谢您的解答，我已经成功地将数据传入，但是在生成mesh的时候又出现了：
Traceback (most recent call last):
  File "demo/run_mapping.py", line 26, in <module>
    slam.run()
  File "/home/nvidia/H2-Mapping/mapping/src/H2Mapping.py", line 26, in run
    self.mapper.run(self.firstframe, update_pose=self.update_pose)
  File "/home/nvidia/H2-Mapping/mapping/src/mapping.py", line 252, in run
    mesh = self.extract_mesh(res=self.mesh_res, clean_mesh=False, map_states=self.map_states)
AttributeError: 'Mapping' object has no attribute 'map_states'
这个问题

JIANG-CX, 2023-10-16:
请问出现mapping start了吗？应该还是和输入的数据有关，请对比我们提供的bag，看下哪里不同。谢谢。

chenghaozhangswu, 2023-10-16:
> 请问出现mapping start了吗？应该还是和输入的数据有关，请对比我们提供的bag，看下哪里不同。谢谢。
> […](#)

出现了mapping start

JIANG-CX, 2023-10-16:
mapping start后有出现数字吗？没有的话就说明输入的数据不对。请确认readme中“use your rgbd sequence”的note都注意到了。同时建议debug step by step看一下和播放bag的运行有什么区别。

获取 Outlook for iOS<https://aka.ms/o0ukef>

出现了mapping start

chenghaozhangswu, 2023-10-16:
好的，没有出现数字

JIANG-CX, 2023-10-16:
嗯，那就说明没有接到合理的数据。

获取 Outlook for iOS<https://aka.ms/o0ukef>

chenghaozhangswu, 2023-10-16:
作者在下面发过了

┢┦aΡｐy笶
***@***.***

lvmingzhe, 2023-10-16:
我是将camera_pose改成cam_pose之后才接收到合理的数据，可以走完实际采集的流程。
另外我想问一下原作，tower场景就两三个小房间，录制了15分钟，拍摄速度十分缓慢，这主要出于什么考虑？是担心L515相机运动模糊还是建图线程处理能力限制？谢谢

JIANG-CX, 2023-10-16:
> 我是将camera_pose改成cam_pose之后才接收到合理的数据，可以走完实际采集的流程。 另外我想问一下原作，tower场景就两三个小房间，录制了15分钟，拍摄速度十分缓慢，这主要出于什么考虑？是担心L515相机运动模糊还是建图线程处理能力限制？谢谢

感谢发现的一个我们清理代码时没检查出的bug😊，因为提供的bag中存在了一个当时实验时记录的camera_pose的topic，所以在dvins代码中为了避免冲突，将topic改成cam_pose了，现已更新，谢谢。
针对你提到的问题，首先L515因为是卷帘快门，运动模糊确实比较严重，同时因为机载电脑上算力相对有限，因此实验时为了让最后建图效果更好，所以选择了较慢的移动。但后续我们将记录下来的中间结果可视化后也发现，其实一些地方很早就已经建好了，是可以走的快一些的hhh。
在我们后续的工作中也会针对建图速度做更进一步的提升，谢谢你的关注和意见。

lvmingzhe, 2023-10-16:
感谢您的神速回复！！我尝试使用了D455全局快门相机来采集数据，但可能是由于rgb/imu标定信息，或者launch文件里对应的json_file_path配置parameters参数没对，目前建图效果比较差。
关于这些参数，您有什么建议吗？
例如
"parameters": {
        "Alternate IR": 0.0,
        "Apd Temperature": -9999,
        "Confidence Threshold": 1,
        "Depth Offset": 4.5,
        "Depth Units": 0.000250000011874363,
        "Digital Gain": 1,
        "Enable IR Reflectivity": 0.0,
        "Enable Max Usable Range": 0.0,
        "Error Polling Enabled": 1,
        "Frames Queue Size": 16,
        "Freefall Detection Enabled": 1,
        "Global Time Enabled": 1,
        "Host Performance": 0.0,
        "Humidity Temperature": 36.1058959960938,
        "Inter Cam Sync Mode": 0.0,
        "Invalidation Bypass": 0.0,
        "LDD temperature": 36.3938369750977,
        "Laser Power": 100,
        "Ma Temperature": 34.7539749145508,
        "Mc Temperature": 35.9714241027832,
        "Min Distance": 250,
        "Noise Estimation": 0.0,
        "Noise Filtering": 6,
        "Post Processing Sharpening": 1,
        "Pre Processing Sharpening": 0.0,
        "Receiver Gain": 9,
        "Sensor Mode": 0.0,
        "Visual Preset": 0.0
    },
或者launch文件里的
<rosparam>
      /camera/motion_module/global_time_enabled: true
      /camera/l500_depth_sensor/global_time_enabled: true
      /camera/rgb_camera/global_time_enabled: true
      
      /camera/rgb_camera/enable_auto_exposure: false
      /camera/rgb_camera/enable_auto_white_balance: false
      
      /camera/rgb_camera/exposure: 100
      /camera/rgb_camera/white_balance: 4500
      /camera/rgb_camera/gain: 4096    
      
      /camera/l500_depth_sensor/min_distance: 100
      /camera/l500_depth_sensor/noise_filtering: 6 

  </rosparam>
我不太清楚这里哪些参数相对重要一点，请给指导一下~谢谢！

JIANG-CX, 2023-10-16:
> 感谢您的神速回复！！我尝试使用了D455全局快门相机来采集数据，但可能是由于rgb/imu标定信息，或者launch文件里对应的json_file_path配置parameters参数没对，目前建图效果比较差。 关于这些参数，您有什么建议吗？ 例如
>
> ```
> "parameters": {
>         "Alternate IR": 0.0,
>         "Apd [...]

首先需要保证tracking部分的准确性，可以先单独跑tracking，看看跟踪精度如何；其次对于455，请确保深度和rgb的对齐的；确保内参正确；对于参数的话，建议打开realsense_viewer来可视化调整，rgb层面比较重要的是“      /camera/rgb_camera/exposure: 100、      /camera/rgb_camera/white_balance: 4500、      /camera/rgb_camera/gain: 4096  ”，另外我没有使用过455，请参阅realsense文档，可能和515的配置会有不同，因为毕竟是两种深度传感器，谢谢。

lvmingzhe, 2023-10-17:
感谢回复，我以后试验一下D455配置，并将可能的参数共享上来。
BTW:
在“use-your-own-rgb-d-sequence”参数配置过程中您提到了offset，bound，num_vertexes三个指标，例如tower场景：
```
offset: 10 # used to make make the coordinate of each point positive
bound: [ [ 0.0,20.0 ],[ 0.0,20.0 ],[ 0.0,10.0 ] ] # [x_min,x_max],[y_min,y_max],[z_min,z_max]
num_vertexes: 200000
```
我的问题是
1. bound的单位是米吗？
2. 内存允许的情况下，num_vertexes是不是尽量大就完事了？对于tower这种场景，200000是上限吗？
3. 为什么要设置offset来保证每个点的坐标值是正的？
4. xyz方向的定义是怎样的？
5. 考虑两种情况：1）从起始点向东运动20米，再向南运动20米，再向上运动10米。2）从起始点向西运动20米，向北运动20米，向下运动10米。这两种情况如何设定bound和offset?
谢谢~

chenghaozhangswu, 2023-10-17:
> 嗯，那就说明没有接到合理的数据。
> […](#)

你好，能否提供你录制bag的realsense sdk和ros版本

JIANG-CX, 2023-10-17:
1.  bound的单位是米吗？
是
  2.  内存允许的情况下，num_vertexes是不是尽量大就完事了？对于tower这种场景，200000是上限吗？
过大的num_vertexes会造成建立计算图时资源浪费，影响速度。200000并不是严格上界。
  3.  为什么要设置offset来保证每个点的坐标值是正的？
因为内部构建八叉树时恒正便于操作
  4.  xyz方向的定义是怎样的？
如果使用的是提供的tracking，就是和vins-fusion的世界系相同。世界坐标系原点就是tracking初始化后的点，右手系，front-left-up，z轴时按估计的重力向上
  5.  考虑两种情况：1）从起始点向东运动20米，再向南运动20米，再向上运动10米。2）从起始点向西运动20米，向北运动20米，向下运动10米。这两种情况如何设定bound和offset?
只和相对于初始化后定义的世界系的方位有关。如果你不确定，你可以把offset和bound点上界都设置的大一些防止出错。

获取 Outlook for iOS<https://aka.ms/o0ukef>

chenghaozhangswu, 2023-10-18:
> 感谢您的神速回复！！我尝试使用了D455全局快门相机来采集数据，但可能是由于rgb/imu标定信息，或者launch文件里对应的json_file_path配置parameters参数没对，目前建图效果比较差。 关于这些参数，您有什么建议吗？ 例如
>
> ```
> "parameters": {
>         "Alternate IR": 0.0,
>         "Apd [...]

我的重建结果也比较模糊

lvmingzhe, 2023-10-18:
能不能把你的重建结果贴上来一起分析一波？
我猜测效果可能和以下配置有关：
1. 位姿估计准确程度
2. 光照条件与对应的rgb_camera/exposure, white_balance, gain配置参数
3. 建图尺寸设置的范围
4. 照片是否有运动模糊
5. 拍摄场景面积与拍摄时长

chenghaozhangswu, 2023-10-18:
> 能不能把你的重建结果贴上来一起分析一波？ 我猜测效果可能和以下配置有关：
> 
> 1. 位姿估计准确程度
> 2. 光照条件与对应的rgb_camera/exposure, white_balance, gain配置参数
> 3. 建图尺寸设置的范围
> 4. 照片是否有运动模糊
> 5. 拍摄场景面积与拍摄时长

[image]

chenghaozhangswu, 2023-10-18:
> 能不能把你的重建结果贴上来一起分析一波？ 我猜测效果可能和以下配置有关：
> 
> 1. 位姿估计准确程度
> 2. 光照条件与对应的rgb_camera/exposure, white_balance, gain配置参数
> 3. 建图尺寸设置的范围
> 4. 照片是否有运动模糊
> 5. 拍摄场景面积与拍摄时长

你的重建结果如何？

lvmingzhe, 2023-10-18:
[image]

chenghaozhangswu, 2023-10-18:
> [image]

跟我的差不多哦，精度不高，能调吗？

lvmingzhe, 2023-10-18:
我是想着先尽量低速拍摄，得到尽量不模糊的图片，和准确的位姿，至于其他参数rgb_camera/exposure, white_balance, gain，还需要根据具体光照环境调节。
如果有什么心得可以在这里交流。

chenghaozhangswu, 2023-10-24:
请问我应该怎么修改，能够实时扫描并看到重建效果

JIANG-CX, 2023-10-24:
因为实时marching cube可视化太慢了，暂时不支持实时查看，谢谢。

获取 Outlook for iOS<https://aka.ms/o0ukef>

zcc00210, 2023-11-01:
请问在进行only mapping的时候，Replace the filename in with the built library src/mapping.py  和下面的那行代码怎么理解呢

JIANG-CX, 2023-11-01:
您好，那个是用于导入八叉树的库

获取 Outlook for iOS<https://aka.ms/o0ukef>

gaolingminhhh, 2023-11-03:
请问您在设置offset为较大的数(例如30 )的时候出现过错误吗?
File "/home/echo/H2-Mapping/mapping/src/functions/render_helpers.py", line 451, in bundle_adjust_frames
    assert (hits.sum() > 0)
AssertionError

gaolingminhhh, 2023-11-03:
请问您在设置offset为较大的数(例如30 )的时候出现过错误吗?
`File "/home/echo/H2-Mapping/mapping/src/functions/render_helpers.py", line 451, in bundle_adjust_frames
assert (hits.sum() > 0)
AssertionError`

JIANG-CX, 2023-11-03:
> 请问您在设置offset为较大的数(例如30 )的时候出现过错误吗? `File "/home/echo/H2-Mapping/mapping/src/functions/render_helpers.py", line 451, in bundle_adjust_frames assert (hits.sum() > 0) AssertionError`

请问是跑replica/room0出错了吗

gaolingminhhh, 2023-11-03:
> > 请问您在设置offset为较大的数(例如30 )的时候出现过错误吗? `File "/home/echo/H2-Mapping/mapping/src/functions/render_helpers.py", line 451, in bundle_adjust_frames assert (hits.sum() > 0) AssertionError`
> 
> 请问是跑replica/room0出错了吗
感谢神速度!
我是在跑自己录制的ros包的时候出现错误的. 想在稍微大一些的场景中运行,不知道该如何设置.
这个是我的H2Mapping运行结果和点云结果
[image]
[image]

gaolingminhhh, 2023-11-03:
我好像知道了,是设置max_depth的值吗?

JIANG-CX, 2023-11-03:
> > > 请问您在设置offset为较大的数(例如30 )的时候出现过错误吗? `File "/home/echo/H2-Mapping/mapping/src/functions/render_helpers.py", line 451, in bundle_adjust_frames assert (hits.sum() > 0) AssertionError`
> > 
> > 
> > 请问是跑replica/room0出错了吗
> > 感谢神速度!
> > 我是在跑自己录制的ros包的时候出现错误的. 想在稍微大一些的场景中运行,不知道该如何设置.
> > 这个是我的H2Mapping运行结果和点云结果
> > [image]
> > [image]

offset的作用是让点云的坐标都是正的，首先需要确认一下30是否足够。其次，看报错是因为没有ray接触到已有的voxel。可以看一下是不是当前帧的点云都大于max_depth了。

small-zeng, 2023-11-15:
你好，请问真实采集数据时，采用的是L515自带IMU还是飞控的IMU呢，这个标定过程有参考手册吗

JIANG-CX, 2023-11-15:
> 你好，请问真实采集数据时，采用的是L515自带IMU还是飞控的IMU呢，这个标定过程有参考手册吗

视频中的手持设备实验（也就是提供的bag）是使用的L515自带的IMU，飞机实验使用的是飞控IMU，标定可以使用kalibr：https://github.com/ethz-asl/kalibr

small-zeng, 2023-11-16:
我用L515采集数据，发现这个不支持640 480 分辨率，您有遇到这个情况麻？我看你的参数是这个分辨率
 roslaunch realsense2_camera rs_aligned_depth_lidar.launch 
... logging to /home/nuc/.ros/log/0fa707be-8464-11ee-ad72-e0e1a99f291f/roslaunch-nuc-NUC12WSKi7-12274.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARN: unrecognized 'param' tag in <include> tag
started roslaunch server http://nuc-NUC12WSKi7:39045/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 400
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 70 lines cut ...]
[ INFO] [1700127580.247421277]: Enable PointCloud: Off
[ INFO] [1700127580.247443274]: Align Depth: On
[ INFO] [1700127580.247475288]: Sync Mode: On
[ INFO] [1700127580.247576935]: Device Sensors: 
[ INFO] [1700127580.313725235]: L500 Depth Sensor was found.
[ INFO] [1700127580.326932079]: RGB Camera was found.
[ INFO] [1700127580.327205044]: Motion Module was found.
[ INFO] [1700127580.327275935]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1700127580.327900607]: num_filters: 1
[ INFO] [1700127580.327929384]: Setting Dynamic reconfig parameters.
[ INFO] [1700127581.542798432]: Done Setting Dynamic reconfig parameters.
[ INFO] [1700127581.543512357]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ WARN] [1700127581.543679290]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: None
[ WARN] [1700127581.543736784]: Using default profile instead.
[ INFO] [1700127581.545707096]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1700127581.545997981]: confidence stream is enabled - width: 640, height: 480, fps: 30, Format: RAW8
[ INFO] [1700127581.548054474]: gyro stream is enabled - fps: 400
[ INFO] [1700127581.548081743]: accel stream is enabled - fps: 400
[ INFO] [1700127581.548108161]: setupPublishers...
[ INFO] [1700127581.550588003]: Expected frequency for depth = 30.00000
[ INFO] [1700127581.568369720]: Expected frequency for color = 30.00000
[ INFO] [1700127581.580516485]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1700127581.592685761]: Expected frequency for confidence = 30.00000
[ INFO] [1700127581.599921865]: Start publisher IMU
[ INFO] [1700127581.600700516]: setupStreams...

small-zeng, 2023-11-16:
我使用的你的Launch config配置，不知道问题出在哪里，是515不支持这个分辨率被马

small-zeng, 2023-11-16:
我使用的你的Launch config配置，不知道问题出在哪里，是515不支持这个分辨率？

closed 2024-01-14 by JIANG-CX

Afterg10w, 2024-12-11:
> 我使用的你的Launch config配置，不知道问题出在哪里，是515不支持这个分辨率？

不知道你有解决了这个问题吗，L515要使用640x480分辨率需要先升级固件