Robotics-STAR-Lab/H2-Mapping #9: 运行bash ros_cmd/run_vins_rgbd.sh，然后播放bag--rosbag play tower_compress.orig.bag，出现问题

chenghaozhangswu, 2023-10-07:
[vins_estimator-1] process has died [pid 43467, exit code -11, cmd /home/nvidia/H2-Mapping/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/nvidia/.ros/log/499c42b8-64cf-11ee-ab81-b48c9d3504a1/vins_estimator-1.log].
log file: /home/nvidia/.ros/log/499c42b8-64cf-11ee-ab81-b48c9d3504a1/vins_estimator-1*.log

JIANG-CX, 2023-10-07:
能提供完整的命令行错误截图吗？

chenghaozhangswu, 2023-10-08:
> 能提供完整的命令行错误截图吗？

网上说可能是opencv和cv_bridge版本冲突，jetpack5.1.2的opencv默认是4.5.4，是怎样改到opencv4.2.0的？

JIANG-CX, 2023-10-08:
应该是cv_bridge的问题，不需要改动opencv的版本。可以如下尝试：
1.卸载原来版本的cv_bridge
```powershell
sudo apt-get remove ros-noetic-cv-bridge
```
2.下载新版本cv_bridge
浏览器打开
[GitHub - ros-perception/vision_opencv​github.com/ros-perception/vision_opencv[image]](https://link.zhihu.com/?target=https%3A//github.com/ros-perception/vision_opencv)
选择noetic对应的cv_bridge，
或者直接，
```text
git clone https://github.com/ros-perception/vision_opencv/tree/noetic 
```
3.修改cv_bridge中的cmakelists.txt
> find_package(OpenCV 4.5.2 REQUIRED) //改成你安装的opencv的版本
4.cmake**单独**编译cv_bridge
在cv_bridge文件夹下，
```text
mkdir build
cd build
cmake ..
make
sudo make install
```
5.在开源代码需要的cmaklist.txt中添加cv_bridge的cmake路径 （loop_fusion和vins_estimator两个子模块的cmakelist.txt开头添加）
```text
set(cv_bridge_DIR /usr/local/share/cv_bridge/cmake)  //在开头

JIANG-CX, 2023-10-10:
Tracking部分的编译和conda无关，请debase后在非conda下编译，谢谢。

获取 Outlook for iOS<https://aka.ms/o0ukef>

我按照你的方法操作后，编译的时候anaconda用的boost1.73.0，然后我安装了boost1.73.0，出现了这个错误
/usr/bin/ld: warning: libboost_filesystem.so.1.71.0, needed by /opt/ros/noetic/lib/libimage_transport.so, may conflict with libboost_filesystem.so.1.73.0
/usr/bin/ld: /usr/local/lib/libcv_bridge.so: undefined reference to boost::re_detail_107300::perl_matcher<char const*, std::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)' /usr/bin/ld: /usr/local/lib/libcv_bridge.so: undefined reference to boost::re_detail_107300::perl_matcher<char const*, std::allocator<boost::sub_match<char const*> >, boost::regex_traits<char, boost::cpp_regex_traits > >::match()'
collect2: error: ld returned 1 exit status

chenghaozhangswu, 2023-10-11:
> Tracking部分的编译和conda无关，请debase后在非conda下编译，谢谢。 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

[image]
你提供的bag运行是这样的吗

JIANG-CX, 2023-10-11:
是的

获取 Outlook for iOS<https://aka.ms/o0ukef>

[image]<https://user-images.githubusercontent.com/95561781/274121250-acafb4be-2a5b-4ba4-a48d-f4bd92e6ea30.png>
你提供的bag运行是这样的吗

chenghaozhangswu, 2023-10-11:
> 是的 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)
> ________________________________ 发件人: chenghaozhangswu ***@***.***> 发送时间: Wednesday, October 11, 2023 10:14:23 AM 收件人: SYSU-STAR/H2-Mapping [...]

哪里可以实时看重建效果呢？

JIANG-CX, 2023-10-11:
可以看下readme，还需要运行mapping

获取 Outlook for iOS<https://aka.ms/o0ukef>

哪里可以实时看重建效果呢？

chenghaozhangswu, 2023-10-11:
> 可以看下readme，还需要运行mapping 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)
> ________________________________ 发件人: chenghaozhangswu ***@***.***> 发送时间: Wednesday, October 11, 2023 10:33:23 AM 收件人: [...]

我是按照流程运行的2，3，4，5，6运行的，但是没有全部播放完bag

JIANG-CX, 2023-10-11:
ctrl c关闭mapping的命令行后，在logs文件夹中的mesh文件夹内会有生成的mesh

获取 Outlook for iOS<https://aka.ms/o0ukef>

chenghaozhangswu, 2023-10-11:
> ctrl c关闭mapping的命令行后，在logs文件夹中的mesh文件夹内会有生成的mesh 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)
> ________________________________ 发件人: chenghaozhangswu ***@***.***> 发送时间: Wednesday, October 11, [...]

ok，谢谢我看到了，请问在运行的时候不能实时看三维重建效果吗？

JIANG-CX, 2023-10-11:
> > ctrl c关闭mapping的命令行后，在logs文件夹中的mesh文件夹内会有生成的mesh 获取 Outlook for iOShttps://aka.ms/o0ukef
> > […](#)
> > ________________________________ 发件人: chenghaozhangswu _**@**_._**> 发送时间: Wednesday, October 11, 2023 10:36:44 AM 收件人: SYSU-STAR/H2-Mapping _**@**_.**_> 抄送: JIANG Chenxing _**@**_._**>; Comment _**@**_.**_> 主题: Re: [SYSU-STAR/H2-Mapping] 运行bash ros_cmd/run_vins_rgbd.sh，然后播放bag--rosbag play tower_compress.orig.bag，出现问题 (Issue #9) 可以看下readme，还需要运行mapping 获取 Outlook for iOShttps://aka.ms/o0ukef …
> > ________________________________ 发件人: chenghaozhangswu @.> 发送时间: Wednesday, October 11, 2023 10:33:23 AM 收件人: SYSU-STAR/H2-Mapping @.> 抄送: JIANG Chenxing @.>; Comment @.> 主题: Re: [SYSU-STAR/H2-Mapping] 运行bash ros_cmd/run_vins_rgbd.sh，然后播放bag--rosbag play tower_compress.orig.bag，出现问题 (Issue #9<#9>) 是的 获取 Outlook for iOShttps://aka.ms/o0ukef …
> > ________________________________ 发件人: chenghaozhangswu @.> 发送时间: Wednesday, October 11, 2023 10:14:23 AM 收件人: SYSU-STAR/H2-Mapping @.> 抄送: JIANG Chenxing @.>; Comment @.> 主题: Re: [SYSU-STAR/H2-Mapping] 运行bash ros_cmd/run_vins_rgbd.sh，然后播放bag--rosbag play tower_compress.orig.bag，出现问题 (Issue #9<#9><#9#9>) Tracking部分的编译和conda无关，请debase后在非conda下编译，谢谢。 获取 Outlook for iOShttps://aka.ms/o0ukef …
> 
> ok，谢谢我看到了，请问在运行的时候不能实时看三维重建效果吗？

暂时没有实现，因为地图是隐式表示的，通过marching cube转换成mesh来需要一定的时间，所以暂时没有实现较为高效的实时可视化

chenghaozhangswu, 2023-10-11:
> > > ctrl c关闭mapping的命令行后，在logs文件夹中的mesh文件夹内会有生成的mesh 获取 Outlook for iOShttps://aka.ms/o0ukef
> > > […](#)
> > > ________________________________ 发件人: chenghaozhangswu _**@**_._**> 发送时间: Wednesday, October 11, 2023 10:36:44 AM 收件人: SYSU-STAR/H2-Mapping _**@**_.**_> 抄送: JIANG Chenxing _**@**_._**>; Comment _**@**_.**_> 主题: Re: [SYSU-STAR/H2-Mapping] 运行bash ros_cmd/run_vins_rgbd.sh，然后播放bag--rosbag play tower_compress.orig.bag，出现问题 (Issue #9) 可以看下readme，还需要运行mapping 获取 Outlook for iOShttps://aka.ms/o0ukef …
> > > ________________________________ 发件人: chenghaozhangswu @.> 发送时间: Wednesday, October 11, 2023 10:33:23 AM 收件人: SYSU-STAR/H2-Mapping @.> 抄送: JIANG Chenxing @.>; Comment @.> 主题: Re: [SYSU-STAR/H2-Mapping] 运行bash ros_cmd/run_vins_rgbd.sh，然后播放bag--rosbag play tower_compress.orig.bag，出现问题 (Issue #9<#9>) 是的 获取 Outlook for iOShttps://aka.ms/o0ukef …
> > > ________________________________ 发件人: chenghaozhangswu @.> 发送时间: Wednesday, October 11, 2023 10:14:23 AM 收件人: SYSU-STAR/H2-Mapping @.> 抄送: JIANG Chenxing @.>; Comment @.> 主题: Re: [SYSU-STAR/H2-Mapping] 运行bash ros_cmd/run_vins_rgbd.sh，然后播放bag--rosbag play tower_compress.orig.bag，出现问题 (Issue #9<#9><#9#9>) Tracking部分的编译和conda无关，请debase后在非conda下编译，谢谢。 获取 Outlook for iOShttps://aka.ms/o0ukef …
> > 
> > 
> > ok，谢谢我看到了，请问在运行的时候不能实时看三维重建效果吗？
> 
> 暂时没有实现，因为地图是隐式表示的，通过marching cube转换成mesh来需要一定的时间，所以暂时没有实现较为高效的实时可视化

好吧，那我搞错了

closed 2023-10-11 by chenghaozhangswu