Robotics-STAR-Lab/H2-Mapping #10: anaconda从yaml文件安装虚拟环境时出现很多ResolvePackageNotFound错误

laliwang, 2023-10-10:
疑似您的yaml文件中各依赖之间版本冲突，如何修改yaml文件才能顺利安装？下面是一些我在纯净环境上安装您的虚拟环境时的anaconda报错截图：
[image]
[image]

JIANG-CX, 2023-10-10:
您好，我进行了一下测试，发现nbformat存在冲突，应该改为5.5.0来适配open3d。我更新了yaml文件，请再次测试。新的yaml文件在我电脑的空白conda环境下可以成功安装，谢谢发现了问题。

laliwang, 2023-10-10:
您好，感谢回复。但是我的问题应该是conda安装dependencies时出现的问题，尚未牵扯到pip安装部分，之前我也从未见过这种情况，请问可能是我的机器上conda的原因吗？或者我能否使用pip来安装以上依赖而非conda，因为conda有时候很好用，有时候又因为不知名原因让人摸不着头脑。谢谢！

JIANG-CX, 2023-10-10:
> 您好，感谢回复。但是我的问题应该是conda安装dependencies时出现的问题，尚未牵扯到pip安装部分，之前我也从未见过这种情况，请问可能是我的机器上conda的原因吗？或者我能否使用pip来安装以上依赖而非conda，因为conda有时候很好用，有时候又因为不知名原因让人摸不着头脑。谢谢！

您好，是可以的，可以尝试一下创建一个空的python3.8的环境，然后将需要的库在该环境下通过pip安装，谢谢。

laliwang, 2023-10-10:
好的感谢！我去尝试一下

closed 2023-10-18 by JIANG-CX