Robotics-STAR-Lab/H2-Mapping #11: The mapping only on scannet datasets aborted at a specific frame

laliwang, 2023-10-12:
When I run the code `python -W ignore demo/run_mapping.py configs/scannet/scene0059.yaml` at [GeForce RTX 3090], I notice that the mapping always interrupts unexpectedly at a certain frame. For example, scene0181 interrupts at 879th, scene 0059 interrupts at 1459th. The screenshots are shown as follows:
[image]

Although the error indicates that libgcc_s.so.1 cannnot be found when pthread_cancel is called, the mapping is not supposed to quit before finished. After I solve the error thrown by libgcc_s.so.1, the mapping still interrupts at the same frame without any errors.
[image]

While for the mapping only on replica dataset, it can perform well untill finished.
[image]

JIANG-CX, 2023-10-12:
Apologies for the oversight. The reason is that the original ScanNet pose in this particular frame is inf. Consequently, I have excluded this invalid pose in the latest version. Thanks.

laliwang, 2023-10-12:
Thank you for your modification, I ignore that the pose is potentially invalid.

JIANG-CX, 2023-10-12:
o(*￣▽￣*)ブ

closed 2023-10-12 by JIANG-CX