Robotics-STAR-Lab/H2-Mapping #12: 请问论文中的FTP(frame processing time)是如何计算的呢？

zjnyly, 2023-10-12:
[image]
我按照指引尝试在Replica数据集上运行，发现每帧图片处理时间比论文中的数据高了一个量级。测试平台为3090，运行时非空载。
[image]

JIANG-CX, 2023-10-12:
论文的测试数据是设备只运行这一个程序测试的结果，如果还运行了别的任务势必会影响这个任务的运行速度，谢谢。

zjnyly, 2023-10-12:
您好，我重新切换了一个空闲的测试环境（硬件为 `Intel(R) Xeon(R) Silver 4310 CPU @ 2.10GHz, NVIDIA A100-SXM4-40GB `)， conda环境按照提供的`h2mapping.yaml`重新搭建，CUDA版本11.3)， 但是依旧无法达到论文中的速度，请问是哪里出了问题，以及论文中测速的方案是什么呢？
[image]

JIANG-CX, 2023-10-12:
我们就是使用python的time.time()进行的统计。因为不同的硬件会有不同，我们并没有在A100上进行过测试。

获取 Outlook for iOS<https://aka.ms/o0ukef>

[image]<https://user-images.githubusercontent.com/56783525/274698044-ebb7970b-9873-4f33-b83a-f1de3ae8894d.png>

zjnyly, 2023-10-12:
好的，十分感谢您的回复

closed 2023-10-12 by zjnyly