Robotics-STAR-Lab/H2-Mapping #14: Error in running tracking code

kshitijgoel007, 2023-11-01:
When running
```bash
bash ros_cmd/run_vins_rgbd.sh
```
I get the following
```
[ WARN] [1698848388.071694649]: image discontinue! reset the feature tracker!
[FATAL] [1698848388.071753592]: ASSERTION FAILED
        file = /opt/ros/noetic/include/ros/publisher.h
        line = 107
        cond = false
        message = 
[FATAL] [1698848388.071765814]: Call to publish() on an invalid Publisher
[FATAL] [1698848388.071777162]: 

[vins_estimator-1] process has died [pid 524308, exit code -5, cmd /home/fractal/workspaces/H2-Mapping/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/fractal/.ros/log/94260b7e-78c1-11ee-a58f-8768d8ec9c53/vins_estimator-1.log].
log file: /home/fractal/.ros/log/94260b7e-78c1-11ee-a58f-8768d8ec9c53/vins_estimator-1*.log
```

Any idea why this might be happening? I get this error at `idx: 237` of the `tower` bag file.

JIANG-CX, 2023-11-01:
Hi, I'm wondering if the mapping module is not executed, would the error still occur?

kshitijgoel007, 2023-11-01:
Hi, yes the error still occurs at the same spot when mapping is not running. I think there is an uninitialized publisher somewhere in your code. Can you check?

kshitijgoel007, 2023-11-01:
Error is probably coming from `vins_estimator` and/or its supporting packages.

JIANG-CX, 2023-11-01:
I reran the tracking module and did not encounter the same error. However, I noticed a warning message in your console stating "image discontinuity! Resetting the feature tracker!" This warning typically occurs when the image stream is not stable. I'm wondering if you are storing the bag file on an HDD. If that's the case, could you please try storing the bag file on an SSD? I suspect that the error might be a result of the unstable data streaming, and using an SSD could potentially help mitigate this issue. Thanks.

closed 2023-11-14 by kshitijgoel007

jiahao-hu, 2023-12-11:
> Error is probably coming from and/or its supporting packages.`vins_estimator`

I have met the same error, did you address it?