Robotics-STAR-Lab/H2-Mapping #15: 在python setup.py install这一步时一直不成功

orientwx, 2023-11-10:
(h2mapping) robot@re:~/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/bindings/torch$ python setup.py install
setup.py:5: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import parse_version
Building PyTorch extension for tiny-cuda-nn version 1.7
Obtained compute capabilities [89] from environment variable TCNN_CUDA_ARCHITECTURES
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Feb__7_19:32:13_PST_2023
Cuda compilation tools, release 12.1, V12.1.66
Build cuda_12.1.r12.1/compiler.32415258_0
Detected CUDA version 12.1
Targeting C++ standard 17
running install
/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  self.initialize_options()
running bdist_egg
running egg_info
writing tinycudann.egg-info/PKG-INFO
writing dependency_links to tinycudann.egg-info/dependency_links.txt
writing top-level names to tinycudann.egg-info/top_level.txt
reading manifest file 'tinycudann.egg-info/SOURCES.txt'
writing manifest file 'tinycudann.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/utils/cpp_extension.py:424: UserWarning: There are no g++ version bounds defined for CUDA version 12.1
  warnings.warn(f'There are no {compiler_name} version bounds defined for CUDA version {cuda_str_version}')
building 'tinycudann_bindings._89_C' extension
creating /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/bindings/torch/build/temp.linux-x86_64-cpython-38
creating /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/bindings/torch/build/temp.linux-x86_64-cpython-38/tinycudann
Emitting ninja build file /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/bindings/torch/build/temp.linux-x86_64-cpython-38/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/10] /usr/local/cuda/bin/nvcc  -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/include -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/dependencies -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/dependencies/cutlass/include -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/dependencies/cutlass/tools/util/include -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/dependencies/fmt/include -I/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/include -I/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/include/TH -I/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/robot/anaconda3/envs/h2mapping/include/python3.8 -c -c /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/src/encoding.cu -o /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/bindings/torch/src/encoding.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -std=c++17 --extended-lambda --expt-relaxed-constexpr -U__CUDA_NO_HALF_OPERATORS__ -U__CUDA_NO_HALF_CONVERSIONS__ -U__CUDA_NO_HALF2_OPERATORS__ -Xcompiler=-Wno-float-conversion -Xcompiler=-fno-strict-aliasing -gencode=arch=compute_89,code=compute_89 -gencode=arch=compute_89,code=sm_89 -DTCNN_PARAMS_UNALIGNED -DTCNN_MIN_GPU_ARCH=89 -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=_89_C -D_GLIBCXX_USE_CXX11_ABI=0
FAILED: /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/bindings/torch/src/encoding.o 
/usr/local/cuda/bin/nvcc  -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/include -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/dependencies -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/dependencies/cutlass/include -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/dependencies/cutlass/tools/util/include -I/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/dependencies/fmt/include -I/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/include -I/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/include/TH -I/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/include/THC -I/usr/local/cuda/include -I/home/robot/anaconda3/envs/h2mapping/include/python3.8 -c -c /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/src/encoding.cu -o /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/bindings/torch/src/encoding.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -std=c++17 --extended-lambda --expt-relaxed-constexpr -U__CUDA_NO_HALF_OPERATORS__ -U__CUDA_NO_HALF_CONVERSIONS__ -U__CUDA_NO_HALF2_OPERATORS__ -Xcompiler=-Wno-float-conversion -Xcompiler=-fno-strict-aliasing -gencode=arch=compute_89,code=compute_89 -gencode=arch=compute_89,code=sm_89 -DTCNN_PARAMS_UNALIGNED -DTCNN_MIN_GPU_ARCH=89 -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=_89_C -D_GLIBCXX_USE_CXX11_ABI=0
In file included from /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/include/tiny-cuda-nn/cpp_api.h:32:0,
                 from /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/include/tiny-cuda-nn/common_host.h:33,
                 from /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/include/tiny-cuda-nn/object.h:32,
                 from /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/include/tiny-cuda-nn/encoding.h:33,
                 from /home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/src/encoding.cu:30:
/home/robot/slam/H2-Mapping/mapping/third_party/tiny-cuda-nn/dependencies/json/json.hpp:3954:14: fatal error: filesystem: 没有那个文件或目录
     #include <filesystem>
              ^~~~~~~~~~~~
compilation terminated.
[... 132 lines cut ...]
  File "/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/setuptools/dist.py", line 989, in run_command
    super().run_command(command)
  File "/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 88, in run
    _build_ext.run(self)
  File "/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 345, in run
    self.build_extensions()
  File "/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 873, in build_extensions
    build_ext.build_extensions(self)
[... cut ...]
  File "/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 686, in unix_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1774, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/home/robot/anaconda3/envs/h2mapping/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 2116, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension

JIANG-CX, 2023-11-10:
It appears that the issue is related to tinycudann. I recommend raising the problem in the repository of tinycudann to seek assistance and resolve the issue.

orientwx, 2023-11-10:
> It appears that the issue is related to tinycudann. I recommend raising the problem in the repository of tinycudann to seek assistance and resolve the issue.

Thank you for your response,I'll give it a try.

closed 2023-11-10 by JIANG-CX