Robotics-STAR-Lab/H2-Mapping #16: Get bad mesh with our L515 data

small-zeng, 2023-11-14:
Thanks for your ncie work.  After I collect data with L515 camera ，we use droid slam to estimate pose and reconstruction the scene with H2-mapping, why we get so bad mesh?

[image]

JIANG-CX, 2023-11-14:
Several factors can influence the final results, such as the accuracy of the pose estimation and the duration of training. I apologize for not being able to directly identify the issue with your scene based on the provided mesh. However, I suggest comparing it to our provided dataset to identify any potential discrepancies or problems. Thank you for your understanding.

small-zeng, 2023-11-15:
> Several factors can influence the final results, such as the accuracy of the pose estimation and the duration of training. I apologize for not being able to directly identify the issue with your [...]

感谢你的神速回复，我看你的论文中提到了coarse tsdf， 我是否可以从代码中导出这个模型(不启动训练)，来验证我的位姿或者图片数据是否有问题

JIANG-CX, 2023-11-15:
可以，你可以试一下把do_mapping函数都注释了，然后得到的mesh近似于coarse tsdf

获取 Outlook for iOS<https://aka.ms/o0ukef>

small-zeng, 2023-11-15:
> 可以，你可以试一下把do_mapping函数都注释了，然后得到的mesh近似于coarse tsdf 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

谢谢解答，我已经注释了do mapping， 得到以下mesh， 似乎很奇怪，噪点很多
[image]

JIANG-CX, 2023-11-15:
请问你的max_depth设置的多少呢？可以试一下设置小一些，因为515过远的深度噪声较大。也可以看一下realsense viewer，是否合理设置了515的参数

获取 Outlook for iOS<https://aka.ms/o0ukef>

[image]<https://user-images.githubusercontent.com/63596020/283014728-6a477c74-9549-4da7-914b-48f71e6c5f1d.png>

small-zeng, 2023-11-15:
max_depth 这个我设置的是3m， 然后515我使用的默认参数 max range 模式， 当我在您的配置中尝试将vox size从0.1m 提高到0.02m， 似乎噪声减少了，这是为什么呢？
[image]

[image]

[image]

JIANG-CX, 2023-11-15:
请问前面那个图中的噪声是在天花板吗？感觉像是天花板处因为视角过大，点云比较稀疏，且有深度噪声导致的。我在使用稀疏深度测量，比如雷达的时候，也会出现类似的情况。因为我设置里voxel内的最少点云数量，voxel size变小，也许就过滤掉了

获取 Outlook for iOS<https://aka.ms/o0ukef>

[image]<https://user-images.githubusercontent.com/63596020/283017604-015054ca-2e43-4b2a-abe7-1f4ed5f10f44.png>

[image]<https://user-images.githubusercontent.com/63596020/283017578-835a6b61-7849-48f4-923f-334e23b134d7.png>

[image]<https://user-images.githubusercontent.com/63596020/283017653-eacf8dee-72f2-45d1-86bd-31689f6a4d01.png>

small-zeng, 2023-11-15:
> 请问前面那个图中的噪声是在天花板吗？感觉像是天花板处因为视角过大，点云比较稀疏，且有深度噪声导致的。我在使用稀疏深度测量，比如雷达的时候，也会出现类似的情况。因为我设置里voxel内的最少点云数量，voxel size变小，也许就过滤掉了 获取 Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

前面那个图是看向墙和地面，应该是因为视角遮挡问题存在一些稀疏点

JIANG-CX, 2023-11-15:
有可能🤔

获取 Outlook for iOS<https://aka.ms/o0ukef>

small-zeng, 2023-11-15:
那还有什么参数可以调整让他变好吗

small-zeng, 2023-11-15:
我感觉这个噪点如果一直存在，后面训练一直不会好。请问您的tsdf 融合的结果是不是没有这么多噪点

JIANG-CX, 2023-11-15:
可以把515对噪声的滤波再开大一点。然后现在这个方案对pose还是比较敏感的，看你第一个mesh，似乎沙发那边呈现了一些pose的问题。可以对比一下提供的bag，看下噪声水平。

获取 Outlook for iOS<https://aka.ms/o0ukef>

small-zeng, 2023-11-15:
好的感谢，请问这个滤波参数对应配置文件的哪个变量呢<br/><br/><br/><html>
<head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
</head>
<body>
<div class="ntes-mailmaster-quote" style="padding-top: 1px; padding-bottom: 1px" >
    <div style=" margin-top: 20px; margin-bottom: 12px; font-size: 14px; line-height: 1.25; color: #89898c; " >---- 回复的原邮件 ----</div>
    <div style=" margin-bottom: 12px; font-size: 13px; line-height: 1.25; color: #2c2c2e; padding: 2px; border-radius: 8px; background-color: #f0f0f0; " >
        <table width="100%" cellpadding="0" cellspacing="10" border="0">
            <tr>
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; white-space: nowrap; " >发件人</td>
                <td valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " ><a class="mail-from" style="color: #1c83eb; text-decoration: none" ***@***.***" >JIANG ***@***.***&gt;</a></td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; white-space: nowrap; " >日期</td>
                <td class="mail-date" valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " >2023年11月15日 14:19</td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; " >收件人</td>
                <td valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " ><a class="mail-to" style="color: #1c83eb; text-decoration: none" ***@***.***" >***@***.***&gt;</a></td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; " >抄送至</td>
                <td valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " ><a class="mail-cc" style="color: #1c83eb; text-decoration: none" ***@***.***" >Jing ***@***.***&gt;</a>、<a class="mail-cc" style="color: #1c83eb; text-decoration: none" ***@***.***" >***@***.***&gt;</a></td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4em; font-size: 13px; line-height: 1.25; color: #89898c; " >主题</td>
                <td class="mail-subject" valign="top" style=" font-size: 13px; line-height: 1.25; color: #2c2c2e; word-break: break-all; " >Re: [SYSU-STAR/H2-Mapping] Get bad mesh with our L515 data  (Issue #16)</td>
            </tr>
        </table>
    </div>
    <div><p></p>
可以把515对噪声的滤波再开大一点。然后现在这个方案对pose还是比较敏感的，看你第一个mesh，似乎沙发那边呈现了一些pose的问题。可以对比一下提供的bag，看下噪声水平。
<br>

<br>
获取 Outlook for iOS&lt;https://aka.ms/o0ukef&gt;
<br>
________________________________
<br>

<br>

<br>
收件人: SYSU-STAR/H2-Mapping ***@***.***&gt;
<br>
抄送: JIANG Chenxing ***@***.***&gt;; Comment ***@***.***&gt;
<br>
<br>

<br>

<br>
<br>

<br>
<br>
<br>
<br>

<script type="application/ld+json">[
{
"target": "https://github.com/SYSU-STAR/H2-Mapping/issues/16#issuecomment-1811872239",
"url": "https://github.com/SYSU-STAR/H2-Mapping/issues/16#issuecomment-1811872239",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script>
</div>
</div>
</body>
</html>

JIANG-CX, 2023-11-15:
Noise Filtering。https://github.com/SYSU-STAR/H2-Mapping/issues/8#issuecomment-1760985423这里是我设置的config文件，可以参考。

small-zeng, 2023-11-15:
明白了，这里的滤波是指在修改L515的launch 文件把

JIANG-CX, 2023-11-15:
嗯，可以尝试一下

获取 Outlook for iOS<https://aka.ms/o0ukef>

closed 2024-01-14 by JIANG-CX