Robotics-STAR-Lab/H2-Mapping #18: eval_color 缺失 data_range参数

lvmingzhe, 2023-11-21:
你好，我在运行评测代码时候遇到以下报错
```
(h2mapping) hello@hello-desktop:~/code/H2-Mapping/mapping$ python eval/eval_color.py $OUTPUT_FOLDER/bak/config.yaml --result_file $OUTPUT_FOLDER
replica
  0%|                                                  | 0/2000 [00:00<?, ?it/s]tiny-cuda-nn warning: GPUMemoryArena: GPU 0 does not support virtual memory. Falling back to regular allocations, which will be larger and can cause occasional stutter.
  0%|                                                  | 0/2000 [00:04<?, ?it/s]
Traceback (most recent call last):
  File "eval/eval_color.py", line 67, in <module>
    ssim, psnr, depth_L1_err = mapper.render_debug_images(tracked_frame)
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/home/hello/code/H2-Mapping/mapping/./src/mapping.py", line 583, in render_debug_images
    ssim, psnr, depth_L1 = self.logger.log_images(ind, rgb, depth, rcolor, rdepth)
  File "/home/hello/code/H2-Mapping/mapping/./src/loggers.py", line 112, in log_images
    ssim = structural_similarity(gt_color_np[(gt_depth_np != 0.0) * (depth_np != 0.0)],
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/site-packages/skimage/metrics/_structural_similarity.py", line 133, in structural_similarity
    ch_result = structural_similarity(im1[_at(ch)],
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/site-packages/skimage/metrics/_structural_similarity.py", line 194, in structural_similarity
    raise ValueError(
ValueError: Since image dtype is floating point, you must specify the data_range parameter. Please read the documentation carefully (including the note). It is recommended that you always specify the data_range anyway.
```
根据chatGPT的解释，这个错误发生在使用 structural_similarity 函数时。当处理浮点数据类型的图像时，structural_similarity 函数需要一个额外的参数 data_range 来指定数据的可能范围。例如，如果图像的数据类型是浮点数并且值在 0.0 到 1.0 之间，那么 data_range 应该设置为 1.0。如果是8位图像，该值应该设置为255。解决这个问题的方法是，在调用 structural_similarity 函数时，添加一个适当的 data_range 参数。
于是我将src/loggers.py里边的ssim末尾添加了一项```data_range=1.0```
```
ssim = structural_similarity(gt_color_np[(gt_depth_np != 0.0) * (depth_np != 0.0)],
                             color_np[(gt_depth_np != 0.0) * (depth_np != 0.0)], channel_axis=-1, data_range=1.0)
```
但是我不确认该data_range值是否正确，请帮忙看一下，谢谢！

lvmingzhe, 2023-11-21:
运行的结果是
```
(h2mapping) hello@hello-desktop:~/code/H2-Mapping/mapping$ python eval/eval_color.py $OUTPUT_FOLDER/bak/config.yaml --result_file $OUTPUT_FOLDER
replica
  0%|                                                                                                                                                                             | 0/2000 [00:00<?, ?it/s]tiny-cuda-nn warning: GPUMemoryArena: GPU 0 does not support virtual memory. Falling back to regular allocations, which will be larger and can cause occasional stutter.
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2000/2000 [52:27<00:00,  1.57s/it]
ssim mean:  0.89849585 ssim std:  0.031417158
psnr mean:  31.691435944012905 psnr std:  1.6842489220753887
depth_L1 mean:  0.004714218 depth_L1 std:  0.0011786523
100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 2000/2000 [00:46<00:00, 43.15it/s]
```
那看来data_range=1.0应该是对的吧。

JIANG-CX, 2023-11-21:
> 运行的结果是
>
> ```
> (h2mapping) hello@hello-desktop:~/code/H2-Mapping/mapping$ python eval/eval_color.py $OUTPUT_FOLDER/bak/config.yaml --result_file $OUTPUT_FOLDER
> replica [...]

应该是的，可能和skimage版本有关，我使用的skimage是0.19.3，data_range是optional，不用一定给出

lvmingzhe, 2023-11-21:
got it, thanks~

closed 2023-11-21 by lvmingzhe