Robotics-STAR-Lab/H2-Mapping #19: 自建数据集eval程序报错

lvmingzhe, 2023-11-22:
您好，我在运行replica的评测程序时候一切正常
```
(h2mapping) hello@hello-desktop:~/code/H2-Mapping/mapping$ OUTPUT_FOLDER=logs/replica/room0/2023-10-08-17-50-05
(h2mapping) hello@hello-desktop:~/code/H2-Mapping/mapping$ python eval/eval_color.py $OUTPUT_FOLDER/bak/config.yaml --result_file $OUTPUT_FOLDER 
```

但是当运行自建数据集的时候
```
(h2mapping) hello@hello-desktop:~/code/H2-Mapping/mapping$ OUTPUT_FOLDER=logs/realsense/l515/2023-11-22-21-26-43
(h2mapping) hello@hello-desktop:~/code/H2-Mapping/mapping$ python eval/eval_color.py $OUTPUT_FOLDER/bak/config.yaml --result_file $OUTPUT_FOLDER 
Traceback (most recent call last):
  File "eval/eval_color.py", line 38, in <module>
    data_stream = get_dataset(args)
  File "/home/hello/code/H2-Mapping/mapping/./src/utils/import_util.py", line 6, in get_dataset
    print(args.dataset)
AttributeError: 'Namespace' object has no attribute 'dataset'
```
可能是什么地方出了问题，请您指点一下，谢谢！

JIANG-CX, 2023-11-22:
如果是自己数据集，需要模仿`mapping/src/dataset/replica.py`创建一个`MY_DATA.py`，然后在相应的config文件里加上一行`dataset: MY_DATA`.

lvmingzhe, 2023-11-22:
但是又有新的错误：
```
(h2mapping) hello@hello-desktop:~/code/H2-Mapping/mapping$ python eval/eval_color.py $OUTPUT_FOLDER/bak/config.yaml --result_file $OUTPUT_FOLDER
realsense
Traceback (most recent call last):
  File "eval/eval_color.py", line 38, in <module>
    data_stream = get_dataset(args)
  File "/home/hello/code/H2-Mapping/mapping/./src/utils/import_util.py", line 8, in get_dataset
    return Dataset.DataLoader(**args.data_specs)
TypeError: __init__() missing 1 required positional argument: 'data_path'
```

lvmingzhe, 2023-11-22:
我现在有点闹不明白自建数据集的*.yaml应该怎么配置了
```
(base) hello@hello-desktop:~/code/H2-Mapping/mapping/configs/l515$ ls
realsense.yaml  tower.yaml
(base) hello@hello-desktop:~/code/H2-Mapping/mapping/configs/l515$ cat tower.yaml 
base_config: configs/l515/realsense.yaml

exp_name: realsense/l515

data_specs:
  max_depth: 3
  use_gt: True # notice

decoder_specs:
  bound: [ [ 0.0,20.0 ],[ 0.0,20.0 ],[ 0.0,20.0 ] ] # [x_min,x_max],[y_min,y_max],[z_min,z_max]
(base) hello@hello-desktop:~/code/H2-Mapping/mapping/configs/l515$ 
```
这里边和标准的文件相比缺少了data_path

lvmingzhe, 2023-11-22:
replica对应配置的是一个文件夹，自己采集的是一个bag
```
/home/hello/code/H2-Mapping/Datasets/L515_20231122_17-25.bag
```

JIANG-CX, 2023-11-22:
> 我需要模仿replica的格式，将bag中的深度图和rgb图提取出来，建一个同样格式的文件夹对吗？

对，你需要把图提取出来，建一个和replica类似的数据集文件夹。data_path的话，tower.yaml因为没有去做evaluation，所以没有加data_path，如果你要做evaluation，就模仿room_0.yaml加一行就行。总结就是，模仿replica的配置hhh。

jiahao-hu, 2023-12-21:
> > 我需要模仿replica的格式，将bag中的深度图和rgb图提取出来，建一个同样格式的文件夹对吗？
> 
> 对，你需要把图提取出来，建一个和replica类似的数据集文件夹。data_path的话，tower.yaml因为没有去做evaluation，所以没有加data_path，如果你要做evaluation，就模仿room_0.yaml加一行就行。总结就是，模仿replica的配置hhh。

您好我想问一下replica数据集里的traj.txt是如何得到的

JIANG-CX, 2023-12-21:
本工作中的replica是从nice-slam中获取的，具体怎么生成可以参考replica数据集的文档

获取 Outlook for iOS<https://aka.ms/o0ukef>

closed 2024-01-14 by JIANG-CX