Robotics-STAR-Lab/H2-Mapping #20: loss曲线抖动幅度太大

small-zeng, 2023-11-29:
当我把loss曲线画出来后，发现所有loss抖动都很大，rgb loss后来还上升了，这个是正常的吗（但是重建效果还可以，是什么原因呢）？下面分别是两组数据：

Replica room0：
[image]

自己采集的L515数据：
[image]
[image]

JIANG-CX, 2023-11-29:
看起来还是正常的，因为这是一个online的incremental的训练过程，所以画出来的loss是当前训练区域的loss情况，会受到如“当前训练视角新区域的占比”以及“当前区域的复杂程度、训练难度”的影响，比如replica可以看到会有上升的情况，主要是因为看到了较多的新场景，所以导致loss变大。此外我发现现在的结果比之前的mesh好很多，请问是做了哪些调整呢？

small-zeng, 2023-11-30:
你好, 最近我在训练replica room0的时候，显存会逐渐增加，mapping 我重新clone 一份还是这样，这个问题出现在哪里呢：
$ python -W ignore demo/run_mapping.py configs/replica/room_0.yaml 
******* initializing first_frame: 0********
mapping started!
mapping frame:  44%|████████████████████████████████████████████████████████████████████▉                                                                                        | 877/1999 [03:46<04:49,  3.88it/s]
Traceback (most recent call last):
  File "demo/run_mapping.py", line 26, in <module>
    slam.run()
  File "/mnt/dataset/zengjing/H2-Mapping/mapping/src/H2Mapping.py", line 26, in run
    self.mapper.run(self.firstframe, update_pose=self.update_pose)
  File "/mnt/dataset/zengjing/H2-Mapping/mapping/src/mapping.py", line 271, in run
    self.mapping_step(frame_id, tracked_frame, update_pose)
  File "/mnt/dataset/zengjing/H2-Mapping/mapping/src/mapping.py", line 183, in mapping_step
    self.do_mapping(tracked_frame=tracked_frame, update_pose=update_pose)
  File "/mnt/dataset/zengjing/H2-Mapping/mapping/src/mapping.py", line 310, in do_mapping
    bundle_adjust_frames(
  File "/mnt/dataset/zengjing/H2-Mapping/mapping/src/functions/render_helpers.py", line 472, in bundle_adjust_frames
    sampled_xyz = ray(rays_o_all[0].unsqueeze(
  File "/mnt/dataset/zengjing/H2-Mapping/mapping/src/functions/render_helpers.py", line 22, in ray
    return ray_start + ray_dir * depths
torch.cuda.OutOfMemoryError: CUDA out of memory. Tried to allocate 64.00 MiB (GPU 0; 23.70 GiB total capacity; 19.74 GiB already allocated; 10.69 MiB free; 21.05 GiB reserved in total by PyTorch) If reserved memory is >> allocated memory try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF
(sdfstudio) zju@zju-FitServer-G4460-V5:/mnt/dataset/zengjing/H2-Mapping/mapping$

JIANG-CX, 2023-11-30:
应该是因为保存的keyframe逐渐增加了，但看起来你是24G显存？应该不会溢出来着

获取 Outlook for iOS<https://aka.ms/o0ukef>

(sdfstudio) ***@***.***:/mnt/dataset/zengjing/H2-Mapping/mapping$

small-zeng, 2023-11-30:
是的，我是3090， 原来都不会增加，最近不知道总是超出

JIANG-CX, 2023-11-30:
可能要回溯一下历史纪录看看哪里不同了hhh

获取 Outlook for iOS<https://aka.ms/o0ukef>

small-zeng, 2023-11-30:
ok，我再试试，这个理论上是初始就分配好显存，还是给根据场景数据会逐渐变化呢

JIANG-CX, 2023-11-30:
keyframe因为也存在cuda上，所以会逐渐增加

获取 Outlook for iOS<https://aka.ms/o0ukef>

small-zeng, 2023-12-14:
请教个问题，如果我一开始拿到的位姿是有噪声的，但是后期由于SLAMF方法
全局BA 位姿变好了，重建质量可以变好吗，我看有个coarse tsdf， 这个会不会有影响

JIANG-CX, 2023-12-14:
暂时没有加入ba后更新地图的功能

获取 Outlook for iOS<https://aka.ms/o0ukef>

closed 2024-01-14 by JIANG-CX