Robotics-STAR-Lab/H2-Mapping #22: Which  built library should Replace the filename in src/mapping.py with ?

SSZ1, 2023-12-22:
Hi I wanna know which library of tiny-cuda-nn should I use to replace the line of torch.classes.load_library("third_party/sparse_octree/build/lib.xxx/svo.xxx.so")?

[image]

Is the libtiny-cuda-nn.a that I sohuld replace with?

Thanks a lot for your attention!

JIANG-CX, 2023-12-22:
Hi, the line of torch.classes.load_library("third_party/sparse_octree/build/lib.xxx/svo.xxx.so") is related to the sparse_octree package located in the third_party folder, rather than tinycudann. For tinycudann, please refer to 'installation 4' in the README for the appropriate instructions. Happy Winter Solstice！Thanks.

获取 Outlook for iOS<https://aka.ms/o0ukef>

image.png (view on web)<https://github.com/SYSU-STAR/H2-Mapping/assets/36888758/2d179918-c74c-4604-95a9-8eaff72dc263>

SSZ1, 2023-12-22:
Thanks a lot!

closed 2023-12-22 by JIANG-CX