Robotics-STAR-Lab/H2-Mapping #25: 使用`D435i`建图， mesh达不到预期效果

1qingfeng-1, 2024-01-10:
## 程序修改

- 替换mapping.py 118行如下，, 修复`conda python3`环境下无法使用cv_bridge

  ```python
  update_pose = True
          color_image = np.frombuffer(color.data, dtype=np.uint8).reshape(color.height, color.width, -1)
          depth_image = np.frombuffer(depth.data, dtype=np.uint16).reshape(depth.height, depth.width, -1)
          depth_image = np.squeeze(depth_image)
  ```

## 参数修改

- 修改`configs/realsense/realsense.yaml`

  ```yaml
  # 通过 rostopic echo /camera/color/camera_info 查询D435i 内参K
  intrinsic: [ 607.9873046875, 607.9636840820312, 318.7456970214844, 248.21688842773438 ]
  
  # GPU内存只有8GB可用，所以batch_size调小，能跑40轮
  N_rays_each: 4096 # mapping's sampling ray
  batch_size: 256
  num_vertexes: 200000
  ```

- 新建`configs/realsense/d435i.yaml`

  ```yaml
  base_config: configs/realsense/realsense.yaml
  
  exp_name: realsense/d435i
  
  data_specs:
    max_depth: 3
    use_gt: True # notice
  
  decoder_specs:
    bound: [ [ 0.0, 10.0 ],[ 0.0, 10.0 ],[ 0.0, 5.0 ] ] # [x_min,x_max],[y_min,y_max],[z_min,z_max]
  ```

- 修改`src/dvins/config/uav2022/uav_nerf.yaml`

  - 通过 rs-enumerate-devices -c 查询参数
  
   ```text
    Intrinsic of "Color" / 640x480 {YUYV/RGB8/BGR8/RGBA8/BGRA8/Y8/Y16}
      Width:        640
      Height:       480
      PPX:          318.745697021484
      PPY:          248.216888427734
      Fx:           607.9873046875
      Fy:           607.963684082031
   ```
  修改替换
  ```yaml
  projection_parameters:
     # fx: 607.987243652344
     # fy: 607.963684082031
     # cx: 422.745697021484
     # cy: 248.216888427734
  
     # 640 x 480  
     fx: 607.9873046875   # FX
     fy: 607.963684082031 # Fy
     cx: 318.745697021484	# cx
     cy: 248.216888427734 # cy
     
     
  ```
  通过 rs-enumerate-devices -c 查询参数
  
  ```text
  Extrinsic from "Color"    To      "Gyro" :
   Rotation Matrix:
     0.999841        -0.0178037        0.00055596    
     0.0178052        0.999837        -0.00294401    
    -0.000503456      0.00295344       0.999996      
  
   Translation Vector: -0.0201705582439899  0.00509376963600516  0.0112307714298368  
  ```
  
  然后是修改旋转平移参数 
  
  ```yaml
  extrinsicRotation: !!opencv-matrix
     rows: 3
     cols: 3
     dt: d
     data: [ 0.999841, -0.0178037, 0.00055596,   
           0.0178052, 0.999837, -0.00294401,    
        -0.000503456, 0.00295344, 0.99999]
  
  #Translation from camera frame to imu frame, imu^T_cam
  extrinsicTranslation: !!opencv-matrix
     rows: 3
     cols: 1
     dt: d
     data: [-0.0201705582439899,  0.00509376963600516,  0.0112307714298368]
  ```

## 运行程序

```bash
python3 -W ignore demo/run_mapping.py configs/realsense/d435i.yaml -run_ros

./ros_cmd/run_vins_rgbd.sh

rosbag play d435i_h2_mapping.bag
```

## 最终效果

[image]

[image]

## 我错过了什么重要的步骤吗

- 需要帮助

- 我的GPU只有8GB RAM

jiahao-hu, 2024-01-10:
您好我想问一下我用realsense-viewer录制的bag文件中的topic并没有：nav_msgs/Odometry类型的消息，这是怎么回事，/vins_estimator/camera_pose没办法替代

jiahao-hu, 2024-01-10:
> ## 程序修改
> * 替换mapping.py 118行如下，, 修复`conda python3`环境下无法使用cv_bridge
>   ```python
>   update_pose = True
>           color_image = np.frombuffer(color.data, dtype=np.uint8).reshape(color.height, color.width, -1)
>           depth_image = np.frombuffer(depth.data, dtype=np.uint16).reshape(depth.height, depth.width, -1)
>           depth_image = np.squeeze(depth_image)
>   ```
> 
> ## 参数修改
> * 修改`configs/realsense/realsense.yaml`
>   ```yaml
>   # 通过 rostopic echo /camera/color/camera_info 查询D435i 内参K
>   intrinsic: [ 607.9873046875, 607.9636840820312, 318.7456970214844, 248.21688842773438 ]
>   
>   # GPU内存只有8GB可用，所以batch_size调小，能跑40轮
>   N_rays_each: 4096 # mapping's sampling ray
>   batch_size: 256
>   num_vertexes: 200000
>   ```
> * 新建`configs/realsense/d435i.yaml`
>   ```yaml
>   base_config: configs/realsense/realsense.yaml
>   
>   exp_name: realsense/d435i
>   
>   data_specs:
>     max_depth: 3
>     use_gt: True # notice
>   
>   decoder_specs:
>     bound: [ [ 0.0, 10.0 ],[ 0.0, 10.0 ],[ 0.0, 5.0 ] ] # [x_min,x_max],[y_min,y_max],[z_min,z_max]
>   ```
> * 修改`src/dvins/config/uav2022/uav_nerf.yaml`
>   
>   * 通过 rs-enumerate-devices -c 查询参数
>   
>   ```
>    Intrinsic of "Color" / 640x480 {YUYV/RGB8/BGR8/RGBA8/BGRA8/Y8/Y16}
>      Width:        640
>      Height:       480
>      PPX:          318.745697021484
>      PPY:          248.216888427734
>      Fx:           607.9873046875
>      Fy:           607.963684082031
>   ```
>   
>   
>       
>         
>       
>   
>         
>       
>   
>       
>     
>   修改替换
>   ```yaml
>   projection_parameters:
>      # fx: 607.987243652344
>      # fy: 607.963684082031
>      # cx: 422.745697021484
>      # cy: 248.216888427734
>   
>      # 640 x 480  
>      fx: 607.9873046875   # FX
>      fy: 607.963684082031 # Fy
>      cx: 318.745697021484	# cx
>      cy: 248.216888427734 # cy
>      
>      
>   ```
>   
>   
>       
>         
>       
>   
>         
>       
>   
>       
>     
>   通过 rs-enumerate-devices -c 查询参数
>   ```
>   Extrinsic from "Color"    To      "Gyro" :
>    Rotation Matrix:
>      0.999841        -0.0178037        0.00055596    
>      0.0178052        0.999837        -0.00294401    
>     -0.000503456      0.00295344       0.999996      
>   
>    Translation Vector: -0.0201705582439899  0.00509376963600516  0.0112307714298368  
>   ```
>   
>   
>       
>         
>       
>   
>         
>       
>   
>       
>     
>   然后是修改旋转平移参数
>   ```yaml
>   extrinsicRotation: !!opencv-matrix
>      rows: 3
>      cols: 3
>      dt: d
>      data: [ 0.999841, -0.0178037, 0.00055596,   
>            0.0178052, 0.999837, -0.00294401,    
>         -0.000503456, 0.00295344, 0.99999]
>   
>   #Translation from camera frame to imu frame, imu^T_cam
>   extrinsicTranslation: !!opencv-matrix
>      rows: 3
>      cols: 1
>      dt: d
>      data: [-0.0201705582439899,  0.00509376963600516,  0.0112307714298368]
>   ```
> 
> ## 运行程序
> ```shell
> python3 -W ignore demo/run_mapping.py configs/realsense/d435i.yaml -run_ros
> 
> ./ros_cmd/run_vins_rgbd.sh
> 
> rosbag play d435i_h2_mapping.bag
> ```
> 
> ## 最终效果
> ![image](https://private-user-images.githubusercontent.com/33478786/295445325-ef6a3cd6-2d3c-47dd-8d2d-6173612e1bea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4ODU4OTAsIm5iZiI6MTcwNDg4NTU5MCwicGF0aCI6Ii8zMzQ3ODc4Ni8yOTU0NDUzMjUtZWY2YTNjZDYtMmQzYy00N2RkLThkMmQtNjE3MzYxMmUxYmVhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEwVDExMTk1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ2ZTA2YzhjOTFjMWRjNWYwMmY3NDRhYTZhYmY3ZjExMmUyNzM0MGNkNjI3MDQ0NDhjZDIyZGM2NzgxYTZhNDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2MeirOY2drJjC8iXyK_E7Xo8IVeytHsGyZshkbkmPns)
> 
> ![image](https://private-user-images.githubusercontent.com/33478786/295445355-bdeddc9b-3bc8-4088-b020-dc1b272166f7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ4ODU4OTAsIm5iZiI6MTcwNDg4NTU5MCwicGF0aCI6Ii8zMzQ3ODc4Ni8yOTU0NDUzNTUtYmRlZGRjOWItM2JjOC00MDg4LWIwMjAtZGMxYjI3MjE2NmY3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTEwVDExMTk1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQyMmYxNWMzNzg2NmU0YWY4YWY0YWVmM2JmMWVhOGI5NTRkMDUxNmJiN2FjN2JmZTMwYmFjMzc4YWRmZjNmYmUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.y2phew6WNb6eTCWtknai1O7EYZozPLTKRjdwzKLE-Gs)
> 
> ## 我错过了什么重要的步骤吗
> * 需要帮助
> * 我的GPU只有8GB RAM

您好你修改realsense.yaml时候没有修改他的话题名吗

jiahao-hu, 2024-01-10:
> > 您好我想问一下我用realsense-viewer录制的bag文件中的topic并没有：nav_msgs/Odometry类型的消息，这是怎么回事，/vins_estimator/camera_pose没办法替代
> 
> realsense只是提供rgbd图片，pose需要跑tracking来得到

您好在运行mapping模块时python3 -W ignore demo/run_mapping.py configs/realsense/d435i.yaml -run_ros需要用到realsense.yaml文件，那这时yaml文件中的/vins_estimator/camera_pose话题是哪来的呢，因为我查看了一下您提供的bag包它就是有这个话题的

JIANG-CX, 2024-01-10:
> 您好在运行mapping模块时python3 -W ignore demo/run_mapping.py configs/realsense/d435i.yaml -run_ros需要用到realsense.yaml文件，那这时yaml文件中的/vins_estimator/camera_pose话题是哪来的呢，因为我查看了一下您提供的bag包它就是有这个话题的

提供的包是因为把当时实验时tracking模块的/vins_estimator/camera_pose一并录了下来。请参看readme的步骤https://github.com/SYSU-STAR/H2-Mapping?tab=readme-ov-file#run-in-ros-full-slam，需要运行tracking模块

jiahao-hu, 2024-01-10:
您好我自己录制了一个bag文件，并新建了test.yaml内容为
base_config: configs/D435i/D435i.yaml

exp_name: D435i/test

data_specs:
  max_depth: 3
  use_gt: True # notice

decoder_specs:
  bound: [ [ 0.0,10.0 ],[ 0.0,10.0 ],[ 0.0,5.0 ] ] # [x_min,x_max],[y_min,y_max],[z_min,z_max]
然后修改realsense.yaml中的相机内参以及图像和深度的topic为
ros_args:
  intrinsic: [ 910.36962890625, 909.457275390625, 646.2952270507812, 371.6678161621094 ] # K[0, 0], K[1, 1], K[0, 2], K[1, 2]
  color_topic: '/device_0/sensor_1/Color_0/image/data'
  depth_topic: '/device_0/sensor_0/Depth_0/image/data'
  pose_topic: /vins_estimator/cam_pose
但是在我运行tracking模块后
mapping进程没有反应
python -W ignore demo/run_mapping.py configs/D435i/test.yaml -run_ros
intrinsic:  [[910.36962891   0.         646.29522705]
 [  0.         909.45727539 371.66781616]
 [  0.           0.           1.        ]]
 ========== MAPPING START ===========

JIANG-CX, 2024-01-10:
> 您好我自己录制了一个bag文件，并新建了test.yaml内容为 base_config: configs/D435i/D435i.yaml
>
> exp_name: D435i/test
>
> data_specs: max_depth: 3 use_gt: True # notice
>
> decoder_specs: bound: [ [ 0.0,10.0 ],[ 0.0,10.0 ],[ [...]

tracking模块正常运行吗

jiahao-hu, 2024-01-10:
tracking模块就是在play录制的bag然后别的看不出来

jiahao-hu, 2024-01-10:
./ros_cmd/run_vins_rgbd.sh是正常运行的，请问是除了这两个地方还需要改别的吗，现在我是只在realsense.yaml里面添加了自己bag的图像和深度topic

JIANG-CX, 2024-01-10:
> tracking模块就是在play录制的bag然后别的看不出来

tracking模块的rviz正常显示了位姿估计对吗，如果是的话，检查一下 yaml中的 color_topic、depth_topic、pose_topic 是否有正常的输出，mapping没有反应的原因一般是没有接收到rgbd和pose

jiahao-hu, 2024-01-10:
tracking模块的rviz并没有正常显示位姿估计，是不是imu的topic也需要添加到某个位置

JIANG-CX, 2024-01-10:
> tracking模块的rviz并没有正常显示位姿估计，是不是imu的topic也需要添加到某个位置

需要修改src/dvins/config/uav2022/uav_nerf.yaml来适配你的传感器

closed 2024-01-16 by JIANG-CX