Robotics-STAR-Lab/H2-Mapping #26: L515重建效果不佳

orientwx, 2024-01-18:
[image][image]
这是建图的正常水平吗，还得有其他原因？
使用的launch和config文件和你们提供的参考是一样的

JIANG-CX, 2024-01-18:
影响建图质量的因素很多，请参看其他issue中的一些回答进行调整 （eg. #8 ， #16 , #25 等等)

JIANG-CX, 2024-01-19:
第一个报错是显示当前帧下没有任何ray和voxel有相交，请确认一下是否按https://github.com/SYSU-STAR/H2-Mapping#use-your-own-rgb-d-sequence 正确设置了offset等参数

closed 2024-01-30 by JIANG-CX

orientwx, 2024-03-15:
像这种线缆细节为什么建图不出来[image][image]

JIANG-CX, 2024-03-15:
> 像这种线缆细节为什么建图不出来![image](https://private-user-images.githubusercontent.com/136042129/313185899-644dedb1-a9d0-4a77-b8ac-d40046b4ea4d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29 [...]

pose和深度的测量，都会影响重建的效果，现在mesh的分辨率是1/80m，如果pose和深度没问题，也可以提升一下mesh分辨率

hesen3, 2024-03-15:
提升分辨率内存容易爆满，是提高这个mesh_res参数吗？

JIANG-CX, 2024-03-15:
对的，你可以改一下代码，比如分批次，然后放到cpu储存

获取 Outlook for iOS<https://aka.ms/o0ukef>

hesen3, 2024-03-16:
> 对的，你可以改一下代码，比如分批次，然后放到cpu储存 获取 Outlook for iOS<https://aka.ms/o0ukef>
> [...](#)

你好，请问这部分修改可以说的详细一下吗？

JIANG-CX, 2024-03-16:
你可以先提高这个mesh_res参数但只重建小一点的区域，这样就可以减少显存消耗。如果问题得以解决，再看下是extract_mesh中哪个部分溢出了，把那部分放到cpu内存。

你好，请问这部分修改可以说的详细一下吗？