Robotics-STAR-Lab/H2-Mapping #27: 运行FULL SLAM系统结束时遇到报错

lvmingzhe, 2024-01-23:
该错误是发生在下达`rosnode kill -a`指令之后，
仅在运行FULL SLAM系统时候才有，在运行纯mapping线程的时候没有报错。
mesh可以提取出来，似乎也不是一个致命的错误。但我有点好奇能不能有什么思路解决这个问题？
```
idx:  21
idx:  22
idx:  23
idx:  24
idx:  25
idx:  26
idx:  27
idx:  28
idx:  29
idx:  30
shutdown request: user request
******* mapping process died *******
********** post-processing 0 steps **********
******* extracting final mesh *******
********** get color from network **********
******* mapping process died *******
Exception ignored in: <function Image.__del__ at 0xfffee2a34af0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 4017, in __del__
    self.tk.call('image', 'delete', self.name)
RuntimeError: main thread is not in main loop
Exception ignored in: <function Variable.__del__ at 0xfffee2a1a1f0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 363, in __del__
    if self._tk.getboolean(self._tk.call("info", "exists", self._name)):
RuntimeError: main thread is not in main loop
Exception ignored in: <function Variable.__del__ at 0xfffee2a1a1f0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 363, in __del__
    if self._tk.getboolean(self._tk.call("info", "exists", self._name)):
RuntimeError: main thread is not in main loop
Exception ignored in: <function Variable.__del__ at 0xfffee2a1a1f0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 363, in __del__
    if self._tk.getboolean(self._tk.call("info", "exists", self._name)):
RuntimeError: main thread is not in main loop
Exception ignored in: <function Variable.__del__ at 0xfffee2a1a1f0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 363, in __del__
    if self._tk.getboolean(self._tk.call("info", "exists", self._name)):
RuntimeError: main thread is not in main loop
Exception ignored in: <function Image.__del__ at 0xfffee2a34af0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 4017, in __del__
    self.tk.call('image', 'delete', self.name)
RuntimeError: main thread is not in main loop
Exception ignored in: <function Variable.__del__ at 0xfffee2a1a1f0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 363, in __del__
    if self._tk.getboolean(self._tk.call("info", "exists", self._name)):
RuntimeError: main thread is not in main loop
Exception ignored in: <function Variable.__del__ at 0xfffee2a1a1f0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 363, in __del__
    if self._tk.getboolean(self._tk.call("info", "exists", self._name)):
RuntimeError: main thread is not in main loop
Exception ignored in: <function Variable.__del__ at 0xfffee2a1a1f0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 363, in __del__
    if self._tk.getboolean(self._tk.call("info", "exists", self._name)):
RuntimeError: main thread is not in main loop
Exception ignored in: <function Variable.__del__ at 0xfffee2a1a1f0>
Traceback (most recent call last):
  File "/home/hello/mambaforge/envs/h2mapping/lib/python3.8/tkinter/__init__.py", line 363, in __del__
    if self._tk.getboolean(self._tk.call("info", "exists", self._name)):
RuntimeError: main thread is not in main loop
Tcl_AsyncDelete: async handler deleted by the wrong thread
Aborted (core dumped)
```
我问了chatGPT, 大概的解释是关闭其他窗口时候发现已经结束。
以下是chatGPT的解释：
错误与 Python 中的 tkinter 库相关，这是一个用于创建图形用户界面的库。具体错误信息说明在对象的析构函数（__del__）中，尝试与图形用户界面的主线程进行交互时出现了问题。报错 "RuntimeError: main thread is not in main loop" 表明主线程不在主循环中，这通常发生在图形界面的事件循环已经结束，但代码仍尝试更新或操作界面元素时。

最后的 "Tcl_AsyncDelete: async handler deleted by the wrong thread" 错误进一步表明了线程之间的交互问题。在使用多线程编程时，尤其是涉及图形用户界面时，不同的线程需要正确地管理和同步，否则可能会导致此类错误。

JIANG-CX, 2024-01-23:
在运行提供的bag时候的报错吗

获取 Outlook for iOS<https://aka.ms/o0ukef>

lvmingzhe, 2024-01-23:
刚才试了一下，运行官方提供的bag结束时不报错。。。

lvmingzhe, 2024-01-23:
定位出问题了，在`/mapping/configs/realsense/realsense.yaml`
最后一段这样设置的话就会报错前边的错误
```
debug_args:
  verbose: True
  mesh_freq: -1
  render_freq: 10
  save_ckpt_freq: -1
  render_res: [ 320, 240 ]
```

默认状态下是
```
debug_args:
  verbose: false
  mesh_freq: -1
  render_freq: -1
  save_ckpt_freq: -1
  render_res: [ 320, 240 ]
```

JIANG-CX, 2024-01-23:
可以在render_debug_images函数中设置断点看下是哪里出的问题

closed 2024-04-20 by JIANG-CX