Robotics-STAR-Lab/H2-Mapping #28: Why is this system vague in modeling small objects' details? Can parameters be adjusted to enhance clarity?

hesen3, 2024-03-08:
log_dir: './logs'
decoder: parallel_hash_net
update_pose: False

criteria:
  rgb_weight: 0.5 # .5
  depth_weight: 1
  sdf_weight: 50000.0
  fs_weight: 10.0
  sdf_truncation: 0.05 # 0.1

decoder_specs:
  voxel_size: 0.1 # same as mapper_specs
  L: 4  # Number of levels
  F_entry: 2  # Number of feature dimensions per entry
  log2_T: 19  # each level's hashmap_size =  F_entry * (2**F_entry)
  b: 2.0  # each level's resolution = N_min * (b**Level)

mapper_specs:
  start_frame: 0
  end_frame: -1
  N_rays_each: 4096 # mapping's sampling ray
  batch_size: 10240
  num_vertexes: 200000
  inflate_margin_ratio: 0.1
  voxel_size: 0.1
  step_size: 0.1
  num_iterations: 8
  max_voxel_hit: 10
  final_iter: 0
  mesh_res: 8
  overlap_th: 0.8
  kf_window_size: 8
  kf_selection_method: "multiple_max_set_coverage" # "random” or “multiple_max_set_coverage”
  kf_selection_random_radio: 0.5   # random keyframe ratio
  insert_method: "intersection" # "naive" or "intersection"
  insert_ratio: 0.85
  offset: 10 # used to make make the coordinate of each point positive
  use_adaptive_ending: True # adaptive iteration

ros_args:
  intrinsic: [ 601.347290039062, 601.343017578125, 329.519226074219, 238.586654663086 ] # K[0, 0], K[1, 1], K[0, 2], K[1, 2]
  color_topic: '/camera/color/image_raw'
  depth_topic: '/camera/aligned_depth_to_color/image_raw'
  pose_topic: /vins_estimator/cam_pose

debug_args:
  verbose: false
  mesh_freq: -1
  render_freq: -1
  save_ckpt_freq: -1
  render_res: [ 320, 240 ]

hesen3, 2024-03-08:
[image]
[image]

JIANG-CX, 2024-03-08:
In addition to the mapping algorithm, the accuracy of pose estimation and the presence of image motion blur also impact the quality of the reconstruction results.

hesen3, 2024-03-08:
Is this the best result achievable with the "tower_compress.bag" provided by you?

JIANG-CX, 2024-03-12:
No, it is training in the mapping process.

Do your systems require pre-training?

closed 2024-04-20 by JIANG-CX