Robotics-STAR-Lab/H2-Mapping #33: Questions about the execution time in the experimental part

Ubee121, 2024-05-29:
Thank you for your outstanding work! I have a small question about the experimental part. When I run the Replica dataset, the resolution is 1200x680, while for ScanNet, it's 640x480. Why is the execution time per frame higher for ScanNet than for Replica?
[image]

JIANG-CX, 2024-05-29:
Thank you for your question. Since we limit the number of samples and do not sample all the pixels during training, differences in image resolution do not affect the training time. In the case of scannet0000, there are more input images, resulting in a larger keyframe set and increased time for the coverage-maximizing keyframe strategy. Additionally, different scenes have varying geometric structures, which affect octree reconstruction and, consequently, the time required for ray casting.

Ubee121, 2024-05-31:
Thank you for your quick response! I have some technical questions to consult you. Have you ever conducted tests using Jetson Nano? I encountered an error with tinycudann when I tried to deploy on Nano, specifically during the cmake --build build --config RelWithDebInfo -j command. Some people suggested that the error might be due to an outdated version of CUDA, but it seems that Nano supports up to JetPack

JIANG-CX, 2024-05-31:
I'm sorry that I have never conducted tests on the Jetson Nano. Could you please provide the error message displayed in the terminal? This will help me determine if I have encountered this issue before.

Ubee121, 2024-06-02:
In the tinycudann code, I saw that removing the -j command seemed to solve the problem, thank you!

closed 2024-06-02 by Ubee121