Robotics-STAR-Lab/H2-Mapping #35: ModuleNotFoundError: No module named 'cv_bridge'

orientwx, 2024-06-05:
我按FAQ中的步骤卸载'cv_bridge'再手动安装后，运行python -W ignore demo/run_mapping.py configs/realsense/tower.yaml -run_ros出现ModuleNotFoundError: No module named 'cv_bridge'
错误[image]

JIANG-CX, 2024-06-05:
可以参考网上的资料如 https://blog.csdn.net/qq_33854260/article/details/121969800

orientwx, 2024-06-05:
还是不行[image]

JIANG-CX, 2024-06-05:
首先需要确保vins能单独跑起来，也就是图2显示的，这边显示的就是cv_bridge从cpp版没有安装正确，可以google一下网上有很多类似的教程。图1是python下的cv_bridge，现在应该是装好了的

orientwx, 2024-06-05:
vins可以跑起来建图程序又不行了[image]

JIANG-CX, 2024-06-05:
> vins可以跑起来建图程序又不行了C1762003-27E4-4A13-A2F9-B28CD47B823A.jpeg (view on web)<https://github.com/SYSU-STAR/H2-Mapping/assets/136042129/8f04709c-bb2b-4062-9b91-c8e4b42ad91d>

emm我暂时没有遇到过类似的问题，可能和版本或者本地环境都有关系，可以再google一下。

获取 Outlook for iOS<https://aka.ms/o0ukef>

closed 2024-06-13 by JIANG-CX