Robotics-STAR-Lab/H2-Mapping #38: Needed help to start H2-Mapping for realsense camera live data. Getting some error

ShreyasKuntnal, 2024-06-29:
I setup everthing according to the setup but failed find right tutorial or steps to launch but tried with these steps.
 1. roscore
 2. roslaunch for realsense camera
 3. roslaunch for vins estimator
 4. the conda activate h2mapping and later ran this comand: `python3 -W ignore demo/run_mapping.py configs/realsense/realsense.yaml -run_ros` this gives output: 
```
 Traceback (most recent call last):
  File "demo/run_mapping.py", line 25, in <module>
    slam = H2Mapping(args)
  File "/home/vt008/H2-Mapping/mapping/src/H2Mapping.py", line 10, in __init__
    self.logger = BasicLogger(args)
  File "/home/vt008/H2-Mapping/mapping/src/loggers.py", line 20, in __init__
    args.log_dir, args.exp_name, self.get_random_time_str())
AttributeError: 'Namespace' object has no attribute 'exp_name'
```

So, Can you please help me on what went wrong here?

JIANG-CX, 2024-06-30:
Hi, based on the step 5 in “self-captured house dataset", you should use "

configs/realsense/tower.yaml".

获取 Outlook for iOS<https://aka.ms/o0ukef>

ShreyasKuntnal, 2024-06-30:
I guess I wasn't clear or I am not getting what I need.
But Here if i used Tower.yaml will I be able to record mapping of my own house with a realsense camera?

Tower.yaml is only for the dataset bag file which we will download from onedrive linka right?

If I need to create a mapping or 3d ply mesh of my home how to do is my question.

I just wanna run the algorithm on laptop with realsense camera connected and create mapping and save the 3d Ply file.

JIANG-CX, 2024-06-30:
You can follow the guidance in "Use your own rgbd sequence". You should first modify the yaml（check both realsense.yaml and tower.yaml）based on the guidance.
Then you launch the realsense camera and vins to verify the tracking module is successfully running.
Then run
conda activate h2mapping
cd H2-Mapping
source devel/setup.bash
cd mapping
python -W ignore demo/run_mapping.py configs/realsense/tower.yaml -run_ros
to start mapping

获取 Outlook for iOS<https://aka.ms/o0ukef>

JIANG-CX, 2024-06-30:
It is suitable for both the recorded bag or online data stream.

获取 Outlook for iOS<https://aka.ms/o0ukef>

ShreyasKuntnal, 2024-06-30:
So may I know how to fix that particular error when we run realsense.yaml file?

JIANG-CX, 2024-06-30:
The realsense.yaml config serves as the base configuration for tower.yaml, like replica.yaml is the base configuration for room.yaml. When you use tower.yaml, it will inherit the content from realsense.yaml, so you should not directly use realsense.yaml.

ShreyasKuntnal, 2024-07-01:
Oh ok, I kept bag file for decompression, Meanwhile, there are no any other issues in realsense camera nor the vins estimator output that I can work on fixing?

JIANG-CX, 2024-07-01:
For RealSense, you can use rqt_image_view to verify whether you have successfully received the RGB and depth images, or you can check the data stream using ROS topics.

Regarding VINS, it appears that the system fails after initialization, but I cannot determine the source of the error based on the console output. You can send me a screenshot or other details, or you can refer to the numerous tutorials available for running VINS-Mono or VINS-Fusion. The main difference in our tracking module is the use of depth images.

ShreyasKuntnal, 2024-07-01:
I have checked on rqt_image_view stream is evident there, and the rostopic are publishing data. But why does dvins stop with red mark and even mapping process too stops.

Terminal outputs:
1.`bash ros_cmd/run_vins_rgbd.sh` :
```
SUMMARY
========

PARAMETERS
 * /pose_graph/config_file: /home/vt008/H2-Ma...
 * /pose_graph/skip_cnt: 0
 * /pose_graph/skip_dis: 0.0
 * /pose_graph/visualization_shift_x: 0
 * /pose_graph/visualization_shift_y: 0
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /vins_estimator/config_file: /home/vt008/H2-Ma...
 * /vins_estimator/vins_folder: /home/vt008/H2-Ma...

NODES
  /
    pose_graph (pose_graph/pose_graph)
    rvizvisualisation (rviz/rviz)
    vins_estimator (vins_estimator/vins_estimator)

ROS_MASTER_URI=http://localhost:11311

process[vins_estimator-1]: started with pid [13082]
process[pose_graph-2]: started with pid [13083]
process[rvizvisualisation-3]: started with pid [13084]
[ INFO] [1719826597.633981517]: init begins
[ INFO] [1719826597.642896800]: Loaded config_file: /home/vt008/H2-Mapping/src/dvins/vins_estimator/../config/uav2022/uav_nerf.yaml
[ INFO] [1719826597.644509147]: Loaded vins_folder: /home/vt008/H2-Mapping/src/dvins/vins_estimator/../config/../
result path /home/jcx/Documents/RAL2023/ros_ws//vins_result_no_loop.csv
/home/jcx/Documents/RAL2023/ros_ws/ not exists, trying to create it 
/home/jcx/Documents/RAL2023 not exists, trying to create it 
/home/jcx/Documents not exists, trying to create it 
/home/jcx not exists, trying to create it 
Failed to create folder /home/jcx 
[ INFO] [1719826597.644564744]: ROW: 480 COL: 640 
[ WARN] [1719826597.644574163]:  fix extrinsic param 
[ INFO] [1719826597.644622850]: Extrinsic_R : 
   0.999969  0.00614047  0.00501018
-0.00594551    0.999258  -0.0380416
-0.00524006   0.0380106    0.999264
[ INFO] [1719826597.644644404]: Extrinsic_T : 
-0.000121045    -0.020129   -0.0219023
[ INFO] [1719826597.644660286]: Synchronized sensors, fix time offset: -0.015
[ INFO] [1719826597.644682183]: reading paramerter of camera /home/vt008/H2-Mapping/src/dvins/vins_estimator/../config/uav2022/uav_nerf.yaml
[ WARN] [1719826597.644818406]: Waiting for image and imu...
[ WARN] [1719826597.761310770]: Shutdown request received.
[ WARN] [1719826597.761367430]: Reason given for shutdown: [[/rvizvisualisation] Reason: new node registered with same name]
[ WARN] [1719826597.883479979]: First_flag
vt008@vt008:~/H2-Mapping$ [rvizvisualisation-3] process has finished cleanly
log file: /home/vt008/.ros/log/2296f416-378c-11ef-b540-1d303349b635/rvizvisualisation-3*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[ INFO] [1719826599.088863106]: IMU excitation not enouth!
[ INFO] [1719826599.188844031]: IMU excitation not enouth!
[ INFO] [1719826599.288031656]: IMU excitation not enouth!
[ INFO] [1719826599.386265137]: IMU excitation not enouth!
[ INFO] [1719826599.487149598]: IMU excitation not enouth!
[ INFO] [1719826599.587953864]: IMU excitation not enouth!
[ INFO] [1719826599.685284805]: IMU excitation not enouth!
[ INFO] [1719826599.784328930]: IMU excitation not enouth!
[ INFO] [1719826599.886503276]: IMU excitation not enouth!
[ INFO] [1719826599.993498191]: IMU excitation not enouth!
[ INFO] [1719826600.089161195]: IMU excitation not enough!
[... 107 lines cut ...]
[solveRelativePose] inlier ratio: 0.971429
unstable features tracking, please slowly move you device!
unstable features tracking, please slowly move you device!
[GlobalSFM::evaluateError] ATE P: 0.001736
[GlobalSFM::evaluateError] ATE D: 0.027621
[GlobalSFM::evaluateError] ATE P: 0.001461
[GlobalSFM::evaluateError] ATE D: 0.012003
Ceres Solver Report: Iterations: 8, Initial cost: 4.575857e+02, Final cost: 2.415200e+02, Termination: CONVERGENCE
[ WARN] [1719826656.939701848]: gyroscope bias initial calibration -0.0102977  -0.020526  0.0191045
[ INFO] [1719826656.940736676]:  result g     9.76009 -0.887042  -9.37926  -2.54988
[ INFO] [1719826656.940815591]:  refine     9.805 -0.891124  -9.42242  -2.56161
R0 from g: 7.98443e-16      5.2145    -105.209
[ INFO] [1719826656.942145757]: g0     -1.11022e-16  8.88178e-16        9.805
[ INFO] [1719826656.942158722]: my R0  -1.5947e-15     4.26672    -100.282
[ INFO] [1719826656.955525438]: Initialization finish!
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/video/src/lkpyramid.cpp:1257: error: (-215:Assertion failed) (npoints = prevPtsMat.checkVector(2, CV_32F, true)) >= 0 in function 'calc'

[vins_estimator-1] process has died [pid 13082, exit code -6, cmd /home/vt008/H2-Mapping/devel/lib/vins_estimator/vins_estimator __name:=vins_estimator __log:=/home/vt008/.ros/log/2296f416-378c-11ef-b540-1d303349b635/vins_estimator-1.log].
log file: /home/vt008/.ros/log/2296f416-378c-11ef-b540-1d303349b635/vins_estimator-1*.log
```

What to do about it?

You asked for a screenshot can you please be more specific so I can post all the screenshots that u need?

ShreyasKuntnal, 2024-07-01:
This is the screenshot of visualisation.
1.For live input
```
python vis/vis_mesh.py \
> $OUTPUT_FOLDER/bak/config.yaml \
> --result_file $OUTPUT_FOLDER \
> -create_mesh

```
[image]

**In case of tower.bag file**

first place when I python script it ends with cuda memory shortage on my RTX 3060 6GB Graphics card:
[image]

In case if I trigger end for the python and let it create a mapping and save it and run visualisation this what it looks like:
[image]

According to visualization example gif there should be something about the roam right?

So can you please help me with solving these issues?

Can you please (I request you to) give answers to these issues in the numbered order?

ShreyasKuntnal, 2024-07-01:
First I did rosnode kill -a.
Closed roscore terminal.
 I opened a fresh terminal. and started roscore.
 
[image]

in new terminal tab, 
cd H2-Mapping/
source devel/setup.bash
bash ros_cmd/run_vins_rgbd.sh

[image]

Note: You can observe there a newline has started but still process is going on in terminal. This behaviour of starting new terminal is new behaviour I don't know whats causing that.

Next in new terminal tab I have started rosbag

[image]

[image]

JIANG-CX, 2024-07-01:
Ok, it seems the tracking module is installed correctly. So if you use your own realsense data stream, you should also see this rviz visualization, otherwise, the tracking module is not running correctly, you need to check whether you modify the config correctly based on the readme or whether you put your camera in front of a textureless area.

Then you can run the mapping module. If your CUDA memory is limited, you can stop the mapping in advance (ctrl c in the console), then you can check the reconstructed mesh in the mapping/logs.

ShreyasKuntnal, 2024-07-01:
so now for the same rosbag i will try running mapping and visualise the same.

For that I have did rosnode kill -a again  and restarted roscore,
Then again started vins and mapping then started rosbag and let it run for 100s and stopped it.
[image]

Then visualized the same this is what it looks like:
[image]

So is the visualisation correct?

JIANG-CX, 2024-07-01:
Do you find a .ply file in "mapping/logs" ?

ShreyasKuntnal, 2024-07-01:
yes

JIANG-CX, 2024-07-01:
> yes

please provide the screen shot for this .ply file.

ShreyasKuntnal, 2024-07-01:
I have updated the above comment with a screenshot is that the right screenshot?

JIANG-CX, 2024-07-01:
You can use the meshlab(https://www.meshlab.net/#download) to open it.

ShreyasKuntnal, 2024-07-01:
Oh my god it opened. So this is how it looks like. 
Qs 1. Is this the highest quality possible? Or else any other things can be tuned?
[image]

Qs2. Before We close this issue can please help finally with realsense camera live input to please??

JIANG-CX, 2024-07-01:
If you do not use 'rosnode kill -a', some programs may not be terminated, which could cause problems.

Answers to your questions:
Q1: You can try increasing the "mesh_res" parameter in realsense.yaml. This parameter represents the resolution of the grid used in marching cubes (0.1m/mesh_res). However, the results may not change significantly. Alternatively, you can use a better GPU and increase the "num_iteration" parameter in realsense.yaml. This parameter determines the length of the mapping time.

Q2: The live output from the realsense camera is exactly the same as using the bag. You must first confirm that the tracking is successful (inspect the similar rviz visualization), then run the mapping module and use "ctrl c" to stop it.

ShreyasKuntnal, 2024-07-01:
So I recorded the using camera of my table this is how it looks, 
[image]

So if I increase mesh res and iteration I would be able to record even more better?
Can you please tell me more on what basis can I increase the mesh res and iterations? As you said 0.1m/mesh_res what does that mean and what are other things that I need to increase to get a good quality output?

JIANG-CX, 2024-07-01:
The quality of this work largely depends on the quality of the depth data and the accuracy of the pose estimation. Based on the results you provided, the depth data may not be great. You might try using more accurate RGB-D sensors like the L515.

You can find the details of the usage of mesh_res and num_iteration in the code. Briefly speaking, mesh_res is used to determine the grid resolution when using the marching cubes algorithm to generate the mesh (you can Google "marching cubes" for more details). num_iteration controls the number of optimization iterations per frame. Since it is a datastream input, a longer training time means more frames might be neglected as new frames arrive after completing the num_iteration iterations. Therefore, if you have a more powerful GPU, you can increase the num_iteration value, as it will run faster and fewer regions will be neglected.

ShreyasKuntnal, 2024-07-01:
Is setting up this repo on jetson the same?
Oh ok Thank you so much for your faster.

JIANG-CX, 2024-07-01:
The same. The provided bag file was recorded during our Jetson experiment.

ShreyasKuntnal, 2024-07-01:
Oh ok thank you so much sir.
Even the tiny cudnn and everyother thing will be the same right?

closed 2024-07-01 by ShreyasKuntnal

JIANG-CX, 2024-07-01:
The same. The thing you need to notice is that you should install pytorch in https://forums.developer.nvidia.com/t/pytorch-for-jetson/72048