Robotics-STAR-Lab/H2-Mapping #39: A little question about eval_recon.py code

Ubee121, 2024-07-03:
I encountered some minor confusion while running your eval code. Could you please clarify whether the completion ratio obtained from the eval_recon.py code represents the actual coverage? When I only ran it on two pieces of data, the coverage rate was 96.9375%, but visually, it didn't seem to reach 50% of the gt_mesh. Thank you for your response!

JIANG-CX, 2024-07-03:
The completion ratio is determined by calculating the ratio of the competition metric values that are lower than 0.05m. Therefore, it can be considered as the coverage metric.

Ubee121, 2024-07-03:
Thanks for your reply! Could you please help me understand why my actual reconstructed mesh looks like the image below, yet the coverage rate is reported as 96.9375%? I would greatly appreciate your insights. Thank you!
[image]
[image]

JIANG-CX, 2024-07-03:
In our evaluation, we assume that all frames are used. Consequently, we cull portions of the ground truth mesh based on the Octree Voxels in our method to exclude regions not visible in the input frames. 
It appears that not all frames were used in this experiment, resulting in significant culling of the ground truth mesh and causing this issue. 
If you prefer not to cull the mesh, you can change the parameter 'only_cal_has_vox' to 'false'.

closed 2024-07-03 by JIANG-CX