Robotics-STAR-Lab/H2-Mapping #41: How to use Intel Realsense calibrated data in this particular usecase?

ShreyasKuntnal, 2024-07-04:
I have completely implemented H2-Mapping on my PC with 4070 GPU. I have calibrated intel realsense color camera. 
I got yaml file which contains these info of imu calibration and camera calibration
[camera_imu_calib3-report-imucam.pdf](https://github.com/user-attachments/files/16094316/camera_imu_calib3-report-imucam.pdf)

Then I processed these camera and imu data to vins format using [kalib2vins](https://github.com/alploskov/kalibr2vins) package.

Finally I got these below data:
cam0.yaml:
[image]

vins_config.yaml:
[image]

So how to use these opencv matrix in the above vins?

ShreyasKuntnal, 2024-07-05:
@JIANG-CX Can you please help with this issue?

JIANG-CX, 2024-07-06:
hi, I am dealing some urgent things currently. I will response your questions next week as soon as possible.

获取 Outlook for iOS<https://aka.ms/o0ukef>

ShreyasKuntnal, 2024-07-15:
Hi, Can you please look into above issue?

JIANG-CX, 2024-07-15:
You can follow https://github.com/HKUST-Aerial-Robotics/VINS-Mono?tab=readme-ov-file#5-run-with-your-device to write your own config, and change the "config_path" in the uav_nerf.launch