Robotics-STAR-Lab/H2-Mapping #44: sparse octree code confusion

whwh747, 2024-07-10:
Thank you for your outstanding work. While reading the sparse voxel octree code, I encountered some confusion. Could you please explain what types of nodes SURFACE and FEATURE represent? Thank you！

JIANG-CX, 2024-07-15:
之前收到了你的邮件，所以我就用中文回答了哈。这一部分主要是参照了vox-fusion的实现。
一个点云所在体素有八个顶点，八叉树的每一个叶子节点对应的是每一个顶点。其中一个体素中xyz最小的那个顶点就是SURFACE，其余的七个是FEATURE，NONLEAF就是八叉树中不是叶子节点的更大的体素。voxfusion中用了get_centres_and_children 这个函数，但在h2mapping中并没有这个函数

whwh747, 2024-07-16:
感谢您的回复！您说八叉树中每一个叶子节点对应的是每一个顶点？我对八叉树的理解是叶子节点代表的是最小的体素，点云应该是在叶子节点代表的体素中的，是我对稀疏体素八叉树的理解有什么问题吗？在八叉树的代码中NONLEAF节点是没有FEATURE节点的对吧，也就意味着NONLEAF不存储任何embedding或者sdf prior，我对这个稀疏体素八叉树的理解不够深入，而且原始代码也没看懂，希望您可以不吝赐教！感谢！

JIANG-CX, 2024-07-16:
你说的没错，但在这里的实现中，是存的顶点。这是因为sdf prior是存在顶点上的，这样更方便。nonleaf节点是不存sdf prior的

获取 Outlook for iOS<https://aka.ms/o0ukef>

whwh747, 2024-07-17:
感谢您的回复！可不可以理解为每一个点都会插到稀疏体素八叉树中最后一层的叶子节点上（它们是一对一的关系还是一对多的关系呢？），然后叶子节点的七个兄弟节点均为FEATURE节点，八个点一起对落入当前叶子节点的点进行三线性插值获取sdf prior，但是对于NONLEAF节点的话没有其七个兄弟节点（FEATURE节点）？我这样理解对吗？

JIANG-CX, 2024-07-17:
你的理解是对的。提到的一对一和一对多有点没理解意思。算法本质上就是每一个点落入一个voxel，然后查询这个voxel的八个顶点进行插值得到sdf prior。只不过在具体代码实现上是存的顶点，因为这样更方便进行算法的操作。

获取 Outlook for iOS<https://aka.ms/o0ukef>

whwh747, 2024-07-18:
非常感谢你及时的回复，很抱歉我在上次的询问中没讲清楚，我的一对一 一对多的意思是，是否会有多个点共享同一个叶节点？我目前的想法是想根据非叶节点构造多分辨率，使用叶子节点以及其父节点分别插值将特征连接起来一起训练，不知道在您看来这个想法在这个稀疏体素八叉树的代码中是否可行呢？
 

喜欢躺平的xlj
***@***.***

 
 获取 Outlook for iOS<" rel="noopener" target="_blank">https://aka.ms/o0ukef&gt;;

JIANG-CX, 2024-07-18:
只要在一个voxel内的点，都是对应同样的叶子结点。现在的代码不能直接使用，因为只维护了叶子结点的特征

获取 Outlook for iOS<https://aka.ms/o0ukef>

 

喜欢躺平的xlj

 

D

whwh747, 2024-07-18:
非常感谢您耐心的回复，祝您科研顺利，生活愉快！
 

喜欢躺平的xlj
***@***.***

 
 
 
 &amp;nbsp; 
 
 
 喜欢躺平的xlj 
 
 
 
 &amp;nbsp; 
 
 
 
 
 ------------------&amp;nbsp;原始邮件&amp;nbsp;------------------ 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 
 获取 Outlook for iOS<" rel="noopener" target="_blank"&gt;https://aka.ms/o0ukef&amp;gt;;

closed 2024-07-21 by JIANG-CX