Robotics-STAR-Lab/H2-Mapping #47: 确认full_slam是否成功运行

zfkdsg, 2024-09-23:
您好作者 我在运行fullslam的时候 
运行结果如下 
终端1  一直运行python -W ignore demo/run_mapping.py configs/realsense/tower.yaml -run_ros
一直输出idx 1到n
终端2 运行bash ros_cmd/run_vins_rgbd.sh 
刚开始运行的时候自动结束了 但是当我播包后 就输出如下:
9191e+02, Termination: CONVERGENCE
[ WARN] [1727061029.575064325]: gyroscope bias initial calibration   0.0126436  0.00239012 -0.00185548
[ INFO] [1727061029.580520744]:  result g     9.59085 0.842009 -9.37091  -1.8605
[ INFO] [1727061029.580697276]:  refine     9.805  0.86081 -9.58015 -1.90205
R0 from g:        0 -5.03665  -101.23
[ INFO] [1727061029.584356444]: g0     -6.93889e-18  2.44249e-15        9.805
[ INFO] [1727061029.584392914]: my R0         0 -4.63562 -103.427
[ INFO] [1727061029.641615060]: Initialization finish!
[ WARN] [1727061354.158361024]: throw img, only should happen at the beginning

zfkdsg, 2024-09-23:
而且我在运行full_slam的时候 运行bash ros_cmd/run_vins_rgbd.sh的时候经常出现
pose_graph-2] process has died [pid 504681, exit code -6, cmd /home/dlut/zfk/H2-Mapping/devel/lib/pose_graph/pose_graph __name:=pose_graph __log:=/home/dlut/.ros/log/e24ce52c-7198-11ef-9ac7-b391ef66e5ba/pose_graph-2.log].
log file: /home/dlut/.ros/log/e24ce52c-7198-11ef-9ac7-b391ef66e5ba/pose_graph-2*.log
这个问题 请问是为什么
谢谢

JIANG-CX, 2024-09-23:
rviz有正常运行吗？如果run mapping的idx一直在输出说明运行是正常的

获取 Outlook for iOS<https://aka.ms/o0ukef>

zfkdsg, 2024-09-23:
正常输出的 谢谢回答

zfkdsg, 2024-09-23:
您好 我还有一个问题 我在运行full slam的时候 运行成功后阅读代码的时候发现 H2 mapping接受了三个topic 但是这三个topic tower_compress.orig.bag包中都有 也就是说 当我仅运行示例代码的时候不需要运行里面的ros_cmd/run_vins_rgbd.sh 文件 之后在运行自己的程序的时候运行再运行slam就好了

JIANG-CX, 2024-09-23:
如果想直接用bag中的pose的话，可以将config中的pose_topic改成/vins_estimator/camera_pose

获取 Outlook for iOS<https://aka.ms/o0ukef>

zfkdsg, 2024-09-24:
感谢回答

zfkdsg, 2024-09-24:
我今天在看代码的过程中 对于八叉数初始化和插入节点的部分 也就是insert函数有些疑惑 想和你进行讨论
首先：这个函数涉及了三层for循环 第一层for循环 是对每个采样点进行循环 
第二层循环 循环八次 我不理解这里面每个点增加一个偏执的作用是什么？ 为什么要增加这七个偏执？

接着进入第三层循环： 
      这层循环是给每一个点放进八叉树中 
     步骤如下：首先 找到这层循环这个采样点（八个点）的位置 找到对应的父亲节点
     之后新建一个tmp 进行初始化 
    之后就要更新Octree里面存在的五个向量 
  1.首先要更新vox_has_value 这里的意思是这个体素里面存在点
   
  2.如果这个点不是增加偏执的点(七个分别在在xyz+1的点) 也就是说就是我们原本的采样点 
     就在children_中进行更新
    children_的含义是 每一行对应一个octannt 每一列存储这个父亲节点的八个孩子的指针
  
   3.之后进入同样的道理（仅对采样点）  更新voxel_ voxel_的含义：这个体素的xyz 坐标 以及这个体素的边长
    4.最后更新feature_  更新feature_的点是八个点都要更新  但是这里feature_的行 里面有效的内容就是那些最小的体素 
    也就是说 features_ 这个向量仅仅在这个体素是最小体素点的时候有值 对应的值就是leaf_index 也就是我们所谓的八个顶点的索引 
  在更新完fetures_的同时 维护了一个叫做frame_voxel_idx的张量 这个张量存储了leaf-index 我们可以通过这个张量之后用于反向索引这个 feaures里面的内容 
   请问我的理解对吗？
   
   我非常想弄清楚里面八叉数各个向量的具体含义 希望得到您的回答

JIANG-CX, 2024-09-24:
这部分代码主要借鉴于vox-fusion的实现
sdf prior储存在八叉树的grid的八个顶点上，所以实际维护的是八叉树的顶点，所以需要增加偏置去遍历。SURFACE对应的是八个顶点中x/y/z最小的那个点，FEATURE对应其他七个顶点。
你对变量的理解基本是正确的。

zfkdsg, 2024-09-24:
也就是说 正常我把采样点(这里是实际存在的点)作为一个基点构建八叉数 之后增加七个偏执 构建另外的七个定点 八个点合成一个八叉数 
之后我们再用假设的采样点 寻找在哪个体素中 找到八叉数点进行三线形插值 是这个道理吗 谢谢

JIANG-CX, 2024-09-24:
从原理上，就是一个采样点的特征是其所在体素的那个顶点通过插值得到的。在实现中，为了储存这些顶点，是把这些顶点储存成了sparse octree voxel

获取 Outlook for iOS<https://aka.ms/o0ukef>

zfkdsg, 2024-09-27:
您好 对于H2 mapping的新视角合成部分在哪里体现的呢
我在阅读代码的时候好像没有阅读到根据 新的位姿渲染生成图片的部分代码

JIANG-CX, 2024-09-27:
新视角的数据来自于nicer-slam，评估方法同样使用eval_color.py

获取 Outlook for iOS<https://aka.ms/o0ukef>

zfkdsg, 2024-09-27:
感谢回复 我现在再去看一下

closed 2024-12-30 by JIANG-CX