Robotics-STAR-Lab/H2-Mapping #50: 评价指标问题

zfkdsg, 2024-12-17:
作者你好 我最近再看你的H2 mapping论文的时候 发现你的论文中 ssim 和psnr 的评价指标出现了interpolate 和 extrapolate两种情况 请问这两个分别是什么啊 区别是什么呢

JIANG-CX, 2024-12-30:
您好，interpolate 是训练视角上的指标，extraopolate 是nicer-slam中使用的，新视角上的指标

zfkdsg, 2025-01-03:
感谢回复 我现在有一个问题 请问为什么H2 mapping会消耗那么多显存呢 我看网络实际就一两层 
我的问题是H2mapping的网络是每一个体素网格都建立一个网络 还是整体就存在一个sdf残差网络
和 颜色网络
还是因为由于大量的体素点导致的显存要求很多

JIANG-CX, 2025-01-03:
整体就一个网络，显存消耗大部份是保存的关键帧多的原因，你可以使用h3mapping中的关键帧策略，会降低显存消耗

获取 Outlook for iOS<https://aka.ms/o0ukef>

zfkdsg, 2025-01-03:
嗯嗯 好的

zfkdsg, 2025-03-17:
作者 也就是说 我们可以把H3 mapping的关键帧选择策略替换到H2 mapping上对吧

zfkdsg, 2025-03-17:
感谢回复

zfkdsg, 2025-03-17:
好的 我接下来尝试一下 请问一下作者· 对于我修改H2 选择策略 我将H3的移动上去 仅仅对关键帧选择的代码进行修改就可以吧 不需要对于体素等内容进行修改是吗

JIANG-CX, 2025-03-17:
需要修改，请仔细对照两份代码进行修改。或者直接使用h3-mapping

zfkdsg, 2025-03-18:
请问作者 仅仅在H2-mapping上使用H3 的关键帧选择策略 会导致效果变差吗

JIANG-CX, 2025-03-18:
不会

获取 Outlook for iOS<https://aka.ms/o0ukef>

zfkdsg, 2025-03-18:
感谢作者耐心回复 祝愿科研生活顺利

zfkdsg, 2025-03-19:
作者您好 我今天仔细阅读了H2 mapping 和 H3mapping的论文 我发现对于关键帧存储策略来说 不考虑Coverage-maximizing Keyframe Selection的情况下 我需要改变的就是需要在 H2mapping里面 加入 我们的Adaptive Keyframe Pruning函数 也就是及时的剔除我们的kf_graph里面不会用到的关键帧  我的理解正确吗

JIANG-CX, 2025-03-19:
对

获取 Outlook for iOS<https://aka.ms/o0ukef>

zfkdsg, 2025-03-25:
作者您好 我对关键帧剔除有一点疑问 我们真正消耗显存的实际上是kf_graph里面存储的关键帧内容 也就是说 当我们使用
  if len(remove_idx_all) != 0:
                    for remove_idx in remove_idx_all:
                        del kf_graph[remove_idx]
                        del kf_seen_voxel[remove_idx]
                        del kf_seen_voxel_num[remove_idx]
                        del kf_svo_idx[remove_idx]
                        del insert_round[remove_idx]
                        del latest_select_round[remove_idx] 
这段代码 从关键帧kf_graph里面删除关键帧（也就是关键帧存储的RGB和位姿信息）之后 就会节省显存了 是这个道理吗

zfkdsg, 2025-03-25:
作者您好 我现在对H2-mapping中曾加了关键帧剔除策略 修改的代码如下：
再 class Mapping中增加：
 # used for adaptive keyframe pruning   这三个变量是用来存储自适应裁剪机制的  也是今天改的
        self.round_num = 0
        self.kf_latest_select_round = []
        self.kf_insert_round = []
再insert_kf中增加
 if self.adaptive_kf_pruning:               #这里很重要 说的是 如果使用自适应关键帧剪枝
            self.kf_insert_round += [self.round_num]     #这里是插入的轮数
            self.kf_latest_select_round += [0]   
修改do_mapping函数中的：
  elif selection_method == 'multiple_max_set_coverage':   #如果大于的话            
            targets, self.kf_unoptimized_voxels, self.kf_optimized_voxels, self.kf_all_voxels,\
            self.round_num, self.kf_latest_select_round,self.kf_insert_round, \
            self.kf_seen_voxel \
            = multiple_max_set_coverage(
                self.kf_graph,
                self.kf_seen_voxel_num,
                self.kf_unoptimized_voxels,
                self.kf_optimized_voxels,
                self.kf_window_size,
                self.kf_svo_idx,
                self.kf_all_voxels,
                self.num_vertexes,
                self.round_num,
                self.kf_latest_select_round,
                self.kf_insert_round,
                self.kf_seen_voxel
                )
并对multiple_max_set_coverage函数进行修改：
def multiple_max_set_coverage(kf_graph, kf_seen_voxel_num, kf_unoptimized_voxels, kf_optimized_voxels,
                              windows_size, kf_svo_idx, kf_all_voxels, num_vertexes,round_num,latest_select_round,insert_round,kf_seen_voxel):
    cnt = 0 #用于跟踪已经选择的关键帧数量
    target_graph = []  #存储最终选定的关键帧集合    #这里是从关键帧中选出我们要优化的关键帧
    
    padded_tensor = pad_sequence(kf_svo_idx, batch_first=True, padding_value=-1)[:, :, 0]   #将一系列长度不同的张量填充到相同长度
    if kf_unoptimized_voxels is None:    #如果没有未优化的体素 也就是说所有体素都被观察到了  #那么就重新设置,把所有体素全部置为为观测值
        kf_unoptimized_voxels = torch.zeros(num_vertexes).cuda().bool() 
        kf_all_voxels = torch.zeros(num_vertexes).cuda().bool()  # All voxels to be optimized
        kf_optimized_voxels = torch.zeros(
            num_vertexes).cuda().bool()  # The voxels seen by the currently selected x keyframes

        kf_seen_voxel_num = torch.tensor(kf_seen_voxel_num)  # (N) 重新开始 这里是当前看到的体素主梁
        kf_unoptimized_voxels[padded_tensor.long() + 1] += True  # 下面就是重新选择能观测到最大体素的关键帧
        kf_unoptimized_voxels[0] = False
        value, index = torch.max(kf_seen_voxel_num, dim=0)  #选择这些帧中能观测到最大的体素的帧 加入当前优化的图片中
       #从下面开始改
        target_graph += [kf_graph[index]]
        kf_unoptimized_voxels[kf_svo_idx[index].long() + 1] *= False  # Empty number 0   更新为观测到的体素
        kf_optimized_voxels[kf_svo_idx[index].long() + 1] += True   #更新观测到的体素
        cnt += 1   #计数器加1
        #下午开始改下面的代码
    while cnt != int(windows_size):
        result_num = kf_unoptimized_voxels[padded_tensor.long() + 1].sum(-1)  # 几个体素尚未优化
        value, index = torch.max(result_num, dim=0)
        if len(latest_select_round) != 0:
            latest_select_round[index] = round_num
        target_graph += [kf_graph[index]]
        kf_unoptimized_voxels[kf_svo_idx[index].long() + 1] *= False
        kf_optimized_voxels[kf_svo_idx[index].long() + 1] += True
        cnt += 1
        #下面是我们要改的内容
        if kf_unoptimized_voxels.any() == False:  # If all are optimized 瞎几把试一试吧 我也不知道对不对 瞎猫碰死耗子
            if len(latest_select_round) != 0:
                latest_select_round_tensor = torch.tensor(latest_select_round).cuda()                #这里是为了记录每个体素最后一次被观测到的轮次
                insert_round_tensor = torch.tensor(insert_round).cuda()
                remove_mask = (insert_round_tensor + 1 <= round_num) & (latest_select_round_tensor + 1 <= round_num)
                remove_idx_all = remove_mask.nonzero()
                # avoid the index change when del
                remove_idx_all = torch.sort(remove_idx_all, descending=True, dim=0)[0]
                if len(remove_idx_all) != 0:
                    for remove_idx in remove_idx_all:
                        del kf_graph[remove_idx]
                        del kf_seen_voxel[remove_idx]
                        del kf_seen_voxel_num[remove_idx]
                        del kf_svo_idx[remove_idx]
                        del insert_round[remove_idx]
                        del latest_select_round[remove_idx] 
                    padded_tensor = pad_sequence(kf_svo_idx, batch_first=True, padding_value=-1)[:, :, 0] 
            kf_all_voxels[padded_tensor.long() + 1] += True
            kf_all_voxels[0] = False  # Empty number 0
            # Unoptimized voxels = all voxels that need to be optimized - \
            # voxels seen by the currently selected x keyframes
            if len(latest_select_round) == 0:
                kf_unoptimized_voxels = kf_all_voxels * ~kf_optimized_voxels
            else:
                kf_unoptimized_voxels = kf_all_voxels # if the removed one has some unique,
            round_num += 1     
        kf_optimized_voxels *= False
    return target_graph, kf_unoptimized_voxels, kf_optimized_voxels, kf_all_voxels, round_num, latest_select_round, insert_round, kf_seen_voxel

请问作者我修改的是否有哪些函数遗漏了
比如说对于cuda的释放 有没有其他的地方释放了cuda存储的frame 而我却没有释放 
感谢回复